openmv/openmv #859: uvc feature in open mv H7

jeronimoagullo, 2020-07-20:
### Steps to reproduce
I have tried to use the camera open mv H7 as an UVC camera. What I want is to obtain the frames in an Android app and then process the frames in the app.
1. I have uploaded the firmware uvc.bin from the folder OPENMV4
2. the camera works properly in Ubuntu using guvcview
3. The camera is recognised in Android system but it doesn't work in an example apps of this github folder: https://github.com/saki4510t/UVCCamera which works properly with other cameras such as an logitech C270

### Describe your setup
I have used the last version of the IDE and the firmware version v3.6.6. I have used windows 10 to run the open mv IDE.

The date release of my open mv H7 board is 10/27/18. I hope you can help me to get running the uvc firmware properly in the Android device (snapdragon 845).

Thank you in advance

iabdalkader, 2020-07-20:
Hi, some software just doesn't work with our UVC implementation, I suspect the RGB565 pixel format is the issue, if that's the case there's not much I can do to fix this we don't support RGB888.

jeronimoagullo, 2020-07-21:
Hello, thank you for your response. In this case, where may I check the supported formats of the UVC implementation?. As I would like to use in a custom app, may I change the format.

kwagyeman, 2020-07-21:
The UVC code is here:

https://github.com/openmv/openmv/blob/master/src/uvc/src/main.c

Please feel free to submit PRs for making it better.

iabdalkader, 2020-07-21:
> Hello, thank you for your response. In this case, where may I check the supported formats of the UVC implementation?. As I would like to use in a custom app, may I change the format.

Supported formats are Grayscale, YUYV and RGB565. This is where the descriptor is created:
https://github.com/openmv/openmv/blob/master/src/uvc/src/usbd_uvc.c#L471

You can change one of the formats there or add a new one, but you'll probably need to make other changes to the code to add a new format. Since our sensor code only reads 2 bytes per pixel, you'd have to also convert whatever pixel format you have to RGB888.