openmv/openmv #862: person detection script gives "'__divmod__' object isn't iterable" error

acutetech, 2020-07-28:
Running tf_person_detection_search_just_centre.py for the first time I got the following error.

The script had run successfully for several seconds.  Running the script again for a few minutes has not given the fault again.

IDE is 2.5.1 (installed today), camera is M7, and I upgraded the camera firmware to "latest" today also.

**********
Detections at [x=0,y=0,w=240,h=240]
unsure = 0.086275
person = 0.890196
no_person = 0.345098
**********
Detections at [x=60,y=60,w=120,h=120]
unsure = 0.027451
person = 0.580392
no_person = 0.713726
1.418736 fps

Traceback (most recent call last):
  File "<stdin>", line 43, in <module>
TypeError: '__divmod__' object isn't iterable
MicroPython v1.12-omv OpenMV v3.6.7 2020-07-20; OPENMV3-STM32F765

acutetech, 2020-07-28:
Hmmm... I am not sure if you are saying this looks like a hardware fault, or a software fault.

In either case, (somewhat off-topic question): is there a general way of catching unexpected errors so the script does not hang up? Something like a try/except that can wrap around the whole script, and force a reset/reboot? Perhaps the WDT watchdog timer?

kwagyeman, 2020-07-28:
It would be a software issue hopefully. Unless you are testing your code in a radiation heavy area.

For your second question, yes, try catch around the whole script. Alternatively, have a different file that calls your business logic code and have all the error handling in that alternative file.

Try not to catch all exceptions however, e.g. if you catch the IDE exception then the IDE cannot stop your script.

iabdalkader, 2020-07-28:
Hi, you mentioned the M7 cam, do you mean the OpenMV-3/F7 cam ? The OMV3/F7 has much less stack memory (4KBs) and Tensorflow micro library requires a lot of stack memory, so this is most likely a stack overflow in the TF library...If that's the case, I don't know if we can fix this as we already reduced the heap (and increased the stack) as much as possible while still being useful for running other scripts.

acutetech, 2020-07-28:
Yes I have the M7 camera. (which I believe is what you refer to as OpenMV-3/F7 and OMV3/F7, though I don't see these terms on your website). I will upgrade, but if certain features are not supported on the M7 then perhaps it would be helpful to flag this.  

Related point: I upgraded the firmware yesterday (now 3.6.7; don't know old version) and started to get new error messages relating to memory size ("Out of fast Frame Buffer Stack Memory..." in an AprilTag app), so again it would be useful to somehow warn users of implications of firmware changes. Is there a change log? Do new features impact memory availability? Is it possible to drop back to an earlier firmware version?

kwagyeman, 2020-07-28:
Yeah, older firmwares are here: https://github.com/openmv/openmv/releases - along with the change log.

I would say you're going to not get the best experience with older hardware. We do try to keep it working. However, once we have something newer we focus our development on that and just trying to keep the old stuff working with whatever compromises we have to make.

Generally, the firmware for the older boards does improve and more features are added. However, sometimes those improvements come at the expense of RAM usage.

iabdalkader, 2020-07-28:
I may be able to move the stack around and this should fix the issue, from testing it seems TF needs at least 10K up to 16K stack to work. Note this may still break in the future as we don't control how TF manages memory, and in fact it seems to need more memory with every release. I will get back to you soon.

acutetech, 2020-07-29:
Thanks Ibrahim - This is not critical for me now - I was just playing with the feature and reported the problem I found in the interests of QA. 

That said, support for FLIR3 resolution could be important in the future.

iabdalkader, 2020-07-30:
This issue is being fixed now, I've been running the person detector for 10 mins with no issues.
>> That said, support for FLIR3 resolution could be important in the future.

If it's compatible with the older socket we should be able to add support for it.

commit db3d53f3 (2020-07-30): Move OpenMV3 stack to ITCM.

* Fixes #862
PR #864 (2020-07-30): Move OpenMV3 stack to ITCM.
* Fixes #862
closed 2020-07-30 by iabdalkader