openmv/openmv #872: H7 and H7 Plus DCMI_FSYNC_PIN #define is incorrect for external trigger of MT9V034.

BruceSHoward, 2020-08-05:
The DCMI_FSYNC_PIN is presently #defined as GPIO_PIN_5, this equates to processor pin PB5.  This is not consistent with the hardware designs for the H7 and H7 Plus.  The DCMI_FSYNC_PIN should be #defined as GPIO_PIN_4, equating to processor pin PB4.

Presently, the DCMI_FSYNC_PIN is stuck high after system initialization.  This continuously triggers the sensor MT9V034 while in triggered mode.

DCMI_FSYNC_PIN should be #defined as GPIO_PIN_4 to be consistent with the hardware design. This allows software control of the MT9V034 external trigger (Exposure) pin.

**IDE version:** 2.5.1.

**Firmware version:** 3.6.7

**Operating system:**Ubuntu 18.04

### For developers

**GCC version:**

**Build target:**H7 and H7 Plus

kwagyeman, 2020-08-05:
@iabdalkader Can you check the Eagle Board file and see what I/O pin is connected to the camera trigger?

@BruceSHoward Things have been working with trigger for a while now. Are you sure this is broken? There's an external FSYNC pin which isn't actually connected to anything I think.

iabdalkader, 2020-08-06:
Yes this is true, the MT9V034 EXPOSURE pin is connected to PB4.

PR #873 (2020-08-12): Fix FSYNC pin.
* Fixes #872
commit 1526328f (2020-08-12): Fix FSYNC pin.

* Fixes #872
closed 2020-08-12 by iabdalkader