openmv/openmv #875: DAC.NORMAL mode does not write buffer only once in dac.write_timed

ajacobs-github, 2020-08-08:
In dac.write_timed with mode=DAC.NORMAL the buffer should be written to the DAC only once, but it looks like the buffer is being written continuously. 

### Steps to reproduce
1.  See attached test code to reproduce
2.  See attached scope trace of DAC with test code 

### Expected behaviour
Buffer should be written to DAC once, not repeatedly

### Actual behaviour
The buffer is written to the DAC continuously.

### Describe your setup
Board: H7
IDE version: 2.6.0
Firmware version: 3.6.7
Windows 10

[image]

Code to reproduce the issue:

# Untitled - By: Alan - Fri Aug 7 2020

import math
import pyb
import wave
from pyb import DAC
from pyb import Pin

# Setup Debug pin
pin1 = Pin('P1', Pin.OUT_PP)
pin1.value(0)

# setup DAC
dac = DAC(pyb.Pin("P6"), bits=8, buffering = None)

peak = 125      # waveform clips when peak > 126
freq_sine = 20

# create a buffer containing a sine-wave
buf_sine = bytearray(100)
for i in range(len(buf_sine)):
    buf_sine[i] = peak + int((peak-1) * math.sin(2 * math.pi * i / len(buf_sine)))

while(True):

    pin1.value(1)
    pyb.delay(10)
    pin1.value(0)
    # output the sine-wave at frequency
    dac.write_timed(buf_sine, freq_sine * len(buf_sine), mode=DAC.NORMAL)

    pyb.delay(1000)

iabdalkader, 2020-08-08:
You're calling `dac.write_timed` in a loop, am I missing something obvious ?

iabdalkader, 2020-08-08:
@ajacobs-github Can you please test the attached firmware and let me know if it fixes the issue ? BTW I actually opened an [issue](https://github.com/micropython/micropython/issues/5619)  about this in MicroPython's repo and I completely forgot about it. I used a workaround for this issue which introduced this bug because the default DMA mode using this workaround is fixed to normal or circular, with this fixs it should work now.
[firmware.zip](https://github.com/openmv/openmv/files/5045680/firmware.zip)

ajacobs-github, 2020-08-08:
@iabdalkader That fixed it.  Thank you for responding and fixing this bug so quickly.  OpenMV and the H7 are terrific.

[image]

PR #876 (2020-08-08): Fix DAC timed write on H7.
* Fixes #875
closed 2020-08-08 by iabdalkader