openmv/openmv #881: ulab integration into openmv

v923z, 2020-08-14:
Hi all,

This is not really an issue, but a discussion starter. In short, I am completely re-writing `ulab`, and I think this would be the time to take the requirements of `openmv` into account. 

The new features that I am working on are 

1. support for tensors of rank 4 (3 would suffice for `openmv`)
2. views of `ndarrays`
3. full broadcasting
4. `numpy` compatibility of syntax
5. customising of included functions and modules

The first three are not cheap in terms of flash space, the last two are pretty trivial. @iabdalkader mentioned that the `openmv` of `micropython` is already somewhat tight, so I would like to ask you to consider the following options. 

a) reducing the tensor dimension to 3 brings considerable savings

b) if the `int8`, and `int16` data types do not make sense on `openmv`, we could remove them. You would probably still need floats, otherwise all the math functions are pretty useless. The upcasting rules could then be the following:

   - uint8 + uint8 -> uint8
   - uint8 + uint16 -> uint16
   - uint16 + uint16 -> uint16
   - uint8 + float -> float
   - uint16 + float -> float

Let me know, if you want to throw out floats, too, and what we should do about the numerical functions that produce floats: https://micropython-ulab.readthedocs.io/en/latest/ulab.html#numerical. Binary operators are relatively expensive, hence, getting rid of some of them would also reduce firmware size: https://github.com/v923z/micropython-ulab/#ndarray E.g., do you need `pow`, or `true_divide`?

c) Think about the functions that you need. A huge number of them are implemented, but many might not make sense on `openmv`. A list can be found under https://github.com/v923z/micropython-ulab/#modules. The `approx` module could probably be completely removed, and large portions of `linalg` could certainly go. I don't know about `filter`, and many of the functions in `vectorise`, e.g., `erf`, `sinh` etc. (https://micropython-ulab.readthedocs.io/en/latest/ulab.html#universal-functions), and the `vectorize` functions might not be too useful: https://micropython-ulab.readthedocs.io/en/latest/ulab.html#vectorising-generic-python-functions. 

d) almost all functions accept generic iterables, as well as `ndarray`s as their arguments. We can shave off flash space by restricting function arguments to `ndarray`s only. You could still work on generic iterables, if you cast them first, i.e., instead of 

```
import ulab as np
np.std(range(10))
```
you would have to write

```
import ulab as np
np.std(ulab.array(range(10)))
```
(This would also require extra RAM for holding `ulab.array(range(10))`.) 

e) if there are functions either from `numpy`, or `scipy` that are not in `ulab` at the moment, but would be beneficial, let me know. I would be more than happy to include them. 

I don't think that I need a definitely answer to the comments above, in fact, my sole purpose is to find out, what kind of customisation the `openmv` community would like to see. Once the customisation options are in place, you can decide yourself how you want to make use of them. 

Zoltán

@kwagyeman

kwagyeman, 2020-08-14:
@v923z

There's a lot of value from making ulab more standard and getting rid of a lot of our custom stuff. Folks commonly ask us to have np support and to do weird things with arrays. So, I was thinking of making ulab support more important.

a) Not needed. We can remove other features on the camera to make ulab more of the main thing. We've done enough feature creep on the OpenMV Cam now where I know what's valuable. I'm going to start focusing on valuable features and removing things which aren't.

If possible, could you make ulab support an accessor method that integorates the underlying data and returns the right value? E.g. we have RGB565 images which can't be put into your current data types along with 1-bit per pixel binary images which also can't be put into ulab. Only grayscale images fit nicely in uint8. I am also thinking of adding more bayer image support and YUV support in the future. I think it would be awesome if ulab is able to handle all of these images types and automatically unpack pixels, perform matrix ops, and then repack pixels.

So, I'm thinking... that you should make the default ndarray support a custom datatype with an unpack/pack method which can grab a value from a 3D index and then pack a value back into a 3D index (h/w/d). The third dimension being the color channel.

Anyway, the above would be in addition to the types above. E.g. you specify the type as what I'm unpacking to, and if no unpack function is given you assume the default 2d accessor method.

b) Let's keep everything. We can just remove unused libraries. We actually have two vector libraries onboard.

c) Let's keep them. I really just want to enable folks to do what they want and not to ask me for crazy things.

d) Again, don't need to save the space, we will remove unused features.

e) Um, I think if you could just help us with tighter integration with ulab that would be great for our image_t data type. Do you need an OpenMV Cam? We can send you an OpenMV Cam H7 and an OpenMV Cam H7 Plus to play around with. Sending PRs to support our image objects fully with all features of ulab would be huge.

v923z, 2020-08-14:
> If possible, could you make ulab support an accessor method that integorates the underlying data and returns the right value? E.g. we have RGB565 images which can't be put into your current data [...]

I haven't thought of this, and I don't actually know how this is done in `numpy`. One question that comes to mind is how upcasting is handled in this case, and in general, what you convert your RGB565 values to. A tensor of rank 3 (width, height, 3), and of type uint8? Because then once the new code is done, you actually need only the unpacking  function, right?

One issue is that sub-classing is highly non-trivial on `micropython`, so if you need this, I would really suggest tight integration into `ulab` itself, but I think that is OK. I am also certain that the `circtuipython` people would be interested in such a thing. Let me get back to this point, once the other details are ironed out. 

I am not sure what to handle binary data. As far as I remember, even in `numpy`, Booleans are `uint8` arrays with a special flag. What I am trying to say is that you probably can't reduce RAM use by a factor of 8, if that is what you are aiming for. On the other hand, if we treat the bit direction as a special axis, and simply iterate over that (that would be the bit-shifting axis), then it is probably doable. I am brainstorming here, so don't take my word too seriously at this point. But I will think about it.

> Ibrahim added basic ulab support, but, he creates very large arrays when doing operations which limit the use. If ulab can work on the image array directly then this is much better. E.g. maybe we can add an .ndarray()/.array() method to our Image class that returns an ndarray object pointing to the underlying data.

To some degree, this is already possible: you can pass a buffer to `ndarray`. Here is an example (line 70, and thereafter): 
https://learn.adafruit.com/ulab-crunch-numbers-fast-with-circuitpython/fft-example-waterfall-spectrum-analyzer 

Of course, this does absolutely no packing/unpacking, and assumes that your data is one of the supported types. 

> So, I'm thinking... that you should make the default ndarray support a custom datatype with an unpack/pack method which can grab a value from a 3D index and then pack a value back into a 3D index (h/w/d). The third dimension being the color channel.

Do you mind, if we bring others into this discussion? I would really like to hear what the adafruit folks say about this. 

> e) Um, I think if you could just help us with tighter integration with ulab that would be great for our image_t data type. Do you need an OpenMV Cam? We can send you an OpenMV Cam H7 and an OpenMV [...]

Thanks, but can we come back to this a bit later? I need a bit of time to polish off the current code, then I would ask people to help with testing. I am hoping to put the new features (higher-dimensional tensors, views and broadcasting) into master within a month, so that is my time frame for picking up extra stuff, like user-defined data types.

kwagyeman, 2020-08-14:
> I haven't thought of this, and I don't actually know how this is done in numpy. One question that comes to mind is how upcasting is handled in this case, and in general, what you convert your RGB565 [...]

Um, I'm saying that you two function pointers to the ndarray object that if populated are used instead to grab/set the data versus default code. This way I can return an ndarray() for any image type and say it's w/h/d deep and then ulab gets/sets values through two function pointers versus using your default code. This allows ulab to directly access RGB565 images and 1-bit per pixel images. The type the function pointer returns would be the uint8/int8/float/etc.

Of course, the above has a speed hit, but, it saves a bunch of RAM and allows ops on large images. This is what our library does. With this change we could remove a lot of our library code that is a duplicate of ulab stuff.

> To some degree, this is already possible: you can pass a buffer to ndarray. Here is an example (line 70, and thereafter):
https://learn.adafruit.com/ulab-crunch-numbers-fast-with-circuitpython/fft-example-waterfall-spectrum-analyzer

Yeah, but, images are huge... so, the user would have to access things row by row. I don't want them to have to do that. They should be able to just write numpy like code. We have support for this right now but it's lame.

> Do you mind, if we bring others into this discussion? I would really like to hear what the adafruit folks say about this.

Sure, I'm 100% pro though moving a lot of basic math features to numpy.

v923z, 2020-08-15:
> Um, I'm saying that you two function pointers to the ndarray object that if populated are used instead to grab/set the data versus default code. This way I can return an ndarray() for any image type [...]

OK, I see now.

> To be clear... all you need do is have support for using those function pointers in all your code. Then, I just have to make a .ndarry() method for my image object that populates them with the right function calls. The user can then do whatever they need with ulab trivially.

I think this would definitely be an exciting feature.

> Of course, the above has a speed hit, but, it saves a bunch of RAM and allows ops on large images. This is what our library does. With this change we could remove a lot of our library code that is a duplicate of ulab stuff.

Can you point me to code that you have, so that I can get an idea of what is required here?

> Yeah, but, images are huge... so, the user would have to access things row by row. I don't want them to have to do that. They should be able to just write numpy like code. We have support for this right now but it's lame.

I believe, this issue is sorted out, once we can move image data into `numpy` arrays directly via passing the pointer. 

I still have to clean up my code a bit. I will ping you, as soon as I am done with that, so that we can discuss this further.

kwagyeman, 2020-08-15:
Okay.

Regarding the code you wanted to see. There's nothing to point to. Users have just asked for weird things which we don't support.

kwagyeman, 2020-09-12:
@v923z Hey, what's the status on this? I have some time right now to add full support to our code base for our image object to be a first class ulab 2d array.

v923z, 2020-09-13:
There is one more function (`argsort`) that I would need to implement, otherwise, I think I am done. Give me a couple of days.

kwagyeman, 2020-09-13:
Okay, let me know when I should get started.

v923z, 2020-09-19:
I would like to come back to my original point under 4, `numpy` compatibility. There are now two kinds of functions in `ulab`, those from `numpy`, and those from `scipy`. In addition to that, you would like to include functions concerning image manipulation. If we remove the current modular structure, how do you want to call the functions that come from `scipy`? If they are at the top level, i.e., if you can call them without having to import a module, that could be confusing, because `micropython` code won't run in CPython.

kwagyeman, 2020-09-20:
I don't have any particular thoughts on including methods that do image manipulation. I'm primarily interested in methods that offer generic math ops on the image structure we have.

> In addition to that, you would like to include functions concerning image manipulation. If we remove the current modular structure, how do you want to call the functions that come from scipy? If [...]

I don't really understand what you are saying here. What I'm interested in is the ulab library being able to take a buffer access method that I can customize for accessing our image arrays in OpenMV.

v923z, 2020-09-20:
> I don't have any particular thoughts on including methods that do image manipulation. I'm primarily interested in methods that offer generic math ops on the image structure we have.

I think this is only half of the equation. You mentioned somewhere earlier that you would like to have some methods that do some sort of image processing, with the `ndarray` as the data holder. I believe, that is a good idea, but that would require that we add methods that are not in the scope of `numpy`, though, `scipy` has such routines: https://scipy-lectures.org/advanced/image_processing/

> I don't really understand what you are saying here. What I'm interested in is the ulab library being able to take a buffer access method that I can customize for accessing our image arrays in OpenMV.

Then I wasn't clear. Here is a pseudo-snippet:

```
import ulab

a = ulab.imread(some_image)
b = ulab.mean(a, axis=2)
c = ulab.sobel(a, ...)
```
`sobel` is in `scipy.ndimage`, not in `numpy`. So, `sobel`, or anything like that shouldn't be at the top level, because that is not `numpy`-compatible. Now, we could just simply declare an `ndimage` sub-module, but unfortunately, that is not the end of the story, because there are already `numpy`-incompatible functions, e.g., the `sosfilt`: https://github.com/v923z/micropython-ulab#filter . 

At the moment, this is not a problem, because you have to import `filter`, if you want to use either `sosfilt`, or `convolve`, and this is explained in the manual. However, if we get rid of the current sub-modules, then we have to somehow organise the functions, so that the syntax doesn't become confusing.

We could, perhaps, declare 3 sub-modules, `numpy`, `scipy`, and `image_manipulation`, or whatever you want to call that (these could be functions that conform to `openmv` only, and not to some already existing CPython library), and for `numpy` and `scipy`, always adhere to the `numpy/scipy` convention. Somewhat annoyingly, the functions in `numpy` and `scipy` are also organised into sub-modules in a rather haphazard manner. E.g., `linalg`, and `fft` are such sub-modules.

kwagyeman, 2020-09-20:
OpenMV already has a pretty extensive vision library. So, we don't need that right now. It would be interesting for ulab to have this. But, it's quite a lot of work that won't add that much value. It's also not going to be so portable. To get speed a lot of our code is optimized for ARM using DSP instructions. Additionally, we work on the direct image object using different code loops per image type.

So, I wouldn't think about doing any the above. Our "image" module also has all this type of stuff. Unless you add a lot of bloat whatever methods you make will not have enough performance. For example, our performance optimizer Larry just finished giving us bicubic and bilinear scaling code:

https://github.com/kwagyeman/openmv/blob/kwabena/larry_scaling_code/src/omv/img/draw.c#L1191

The code is ugly as sin... but, extremely fast because he did everything using integers without floating point. Writing code this way is pretty much required to have any performance for images since the arrays are so big.

What I'm looking for is just to have the ability to use ulab to do basic 2D array ops on images. E.g. folks have asked me things like:

"I want to multiply the green channel by 2 and then subtract the red and blue channels"

And then they post numpy code showing off how to do that.

It's not possible or sensible for me to include all this is in the OpenMV image library. So, I'm hoping numpy can pick up random things like this that users want.

v923z, 2020-09-20:
> What I'm looking for is just to have the ability to use ulab to do basic 2D array ops on images. E.g. folks have asked me things like:
>
> "I want to multiply the green channel by 2 and then subtract [...]

Oh, then I misunderstood something earlier. So, is it correct that you simply need 

```
import ulab
a = ulab.ndarray(image_buffer).reshape((x, y, 3))
b = 2*a[:,:,1] - a[:,:,0] - a[:,:,2]
c = b.data()
```
i.e., you pass me your image buffer, I do something a la `numpy`, and then I give you a data pointer? This would definitely simplify matters.

kwagyeman, 2020-09-20:
Yes, but, you shouldn't have to know what the OpenMV Cam image object looks like. Instead. I can return an ndarray() that points to data that's inside an image.

```
import ulab, image, sensor
img = sensor.snapshot()
a = img.ndarrary()
b = 2*a[:,:,1] - a[:,:,0] - a[:,:,2]
c = b.data()
```

We have this right now for bytearray() access. E.g. if you want raw byte access to an image object you just do .bytearray() and this creates a small bytearray() object that points to an image data array and allows buffer access to the image array.

So, similarly, if you can create an ndarray() type that doesn't hold it's own data but takes a pointer to a data array, that array size in w/h, and then accessor methods for reading/writing data in that array then I can just implement these methods and create this object which will let ulab access an iamge.

Please keep in mind images are huge. You have to avoid copying them. So, it's quite important that I can create this small class object that allows raw data access. Even if it's not the fastest data access it lets users do anything they want.

...

As for the underlying array type of the data access object... I can make the image appear as an array of uint8/floats, or whatever seems good.

kwagyeman, 2020-09-20:
I don't know how ulab implements:

`a = 2*a[:,:,1] - a[:,:,0] - a[:,:,2]`

But, you can either do the operation element by element which is fine or cache a line of array values and then do the operation line by line which will be much faster. I can implement whatever read/write method versions you need.

What you cannot do is allocate the arrays all on the heap and do the operation in one go.

v923z, 2020-09-20:
> Yes, but, you shouldn't have to know what the OpenMV Cam image object looks like. Instead. I can return an ndarray() that points to data that's inside an image.
> 
> ```
> import ulab, image, sensor
> img = sensor.snapshot()
> a = img.ndarrary()
> a = 2*a[:,:,1] - a[:,:,0] - a[:,:,2]
> # data in the image has been modified now.
> ```
> 
> We have this right now for bytearray() access. E.g. if you want raw byte access to an image object you just do .bytearray() and this creates a small bytearray() object that points to an image data array and allows buffer access to the image array.

I think, `numpy`'s `ndarray` does that, and that should be implemented in `ulab`, so you shouldn't have to implement your method for `img`: `ndarray` can take an arbitrary buffer, and attach to it a small header with the size/strides description of the array. 

> As for the underlying array type of the data access object... I can make the image appear as an array of uint8/floats, or whatever seems good.

This would not be necessary, since you can pass `dtype` to `ndarray`. 

> I don't know how ulab implements:

```
a = 2*a[:,:,1] - a[:,:,0] - a[:,:,2]
```

`a[:,:,1]` etc. are just views of `a`, so no copy is created, because the underlying data pointer is shared. Since we want to do

```
a = ulab.ndarray(image_buffer)
```
which doesn't create new data either, I believe, we can avoid the problem with the image size; we would really be working with what `sensor.snapshot()` returns, nothing more. 

> What you cannot do is allocate the arrays all on the heap and do the operation in one go.

Right. And exactly this is, why I set out to re-write the whole library from scratch, because the old version didn't support views, and that was a serious limitation in this regard.

kwagyeman, 2020-09-20:
Okay, but, I think the image object should return the ndarray(). That way your code doesn't have to known about OpenMV code. So, I need a class object that I can return that acts like an ndarrary().

See the bytearray() object in MP to see what I'm talking about.

v923z, 2020-09-20:
> Okay, but, I think the image object should return the ndarray(). That way your code doesn't have to known about OpenMV code. 

This is true, but you are not the only one, who wants to turn a buffer object into an ndarray, so the `ndarray` method has to be implemented, independent of `openmv`.

kwagyeman, 2020-09-20:
Sure, but, I think this means you need to define a class that others implement and fill in the buffer access.

The image object does allow buffer protocol access. However, you can’t just assume it’s an array of bytes. It could be a binary image (1-bit per pixel), a grayscale image, or an rgb565 image, or a bayer image, or a yuv422 image. So, your code shouldn’t be looking at an image object. But, instead you provide an object that I can implement with the correct buffer protocol access code for the image object.

kwagyeman, 2020-09-20:
What I’m trying to get at is that the image object is a complex data type that you cannot just blindly index.

v923z, 2020-09-21:
> What I’m trying to get at is that the image object is a complex data type that you cannot just blindly index. 

I think this was the missing link. In my understanding, that is. 

> I don’t expect you to support every type of image that could be in my image object. 

The snag is, sub-classing on the C level is not trivial at all, in fact, even in the latest `micropython`, the implementation is somewhat hackish. I have read somewhere that a proper implementation is relatively high on the priority list, but it is not done yet. 

So, I think your best bet is to just "extend" `ulab` with your code, as you would e.g., in the `user` sub-module: https://github.com/v923z/micropython-ulab/#user . Through that, you have access to all `ulab` functions and features, but you can't just define your own class on top of an `ndarray`. 

> So, I just need you to create an ndarray class that allows me to customize how data is read/written.

What do you need for that? It is a bit too abstract for me at the moment. This is what I have now:
```
typedef struct _ndarray_obj_t {
    mp_obj_base_t base;
    uint8_t dense;
    uint8_t dtype;
    uint8_t itemsize;
    uint8_t boolean;
    uint8_t ndim;
    size_t len;
    size_t shape[ULAB_MAX_DIMS];
    int32_t strides[ULAB_MAX_DIMS];
    void *array;
} ndarray_obj_t;
```

We can add additional members to this structure, but we can't really change the fact that `void *array` holds the data.

kwagyeman, 2020-09-21:
Yeah,

I'm saying you should extend the above and add two (or more if you need it) function pointers:

```
typedef struct _ndarray_obj_t {
    mp_obj_base_t base;
    uint8_t dense;
    uint8_t dtype;
    uint8_t itemsize;
    uint8_t boolean;
    uint8_t ndim;
    size_t len;
    size_t shape[ULAB_MAX_DIMS];
    int32_t strides[ULAB_MAX_DIMS];
    void *array;
    (uint32_t read_array)(ndarray_obj_t *self);
    (void write_array)(ndarray_obj_t *self, uint32_t value);
} ndarray_obj_t;
```

So, in my image object I'd return an ndarray() object with all the above information filled out how I want along with a pointer to a read/write function that is different per image type I support. These read/write functions then will do the necessary magic to access the array in a way that makes sense.

Then in your code... if you see that read/write are not zero (i.e. valid function pointers), then you call them to access the data instead of manually accessing in your code.

This then allows you do support whatever user datatype they can think of. For example, we have 1-bit per pixel images. My read/write functions would just pack/unpack bits for access.

v923z, 2020-09-21:
> Yeah,
> 
> I'm saying you should extend the above and add two (or more if you need it) function pointers:

We can definitely add the two function pointers, the question is, what exactly should happen with them. At the moment, you gain speed in `ulab` by reading out the `dtype` member, casting the `void *array` to the indicated type, and then writing out the C loop for that particular type. This means that in most cases, there is no function dispatch in the loops. Here is a simple example: 
https://github.com/v923z/micropython-ulab/blob/c63144397194943b264738025f65706aa64645ae/code/numerical/numerical.c#L116, and https://github.com/v923z/micropython-ulab/blob/c63144397194943b264738025f65706aa64645ae/code/numerical/numerical.h#L55, https://github.com/v923z/micropython-ulab/blob/c63144397194943b264738025f65706aa64645ae/code/numerical/numerical.h#L291

What we could do is to add the extra loops just for `openmv`, and call the function pointer, where we read out the data. It is not going to be fast, though, I am afraid.

kwagyeman, 2020-09-21:
Yeah, for grayscale images things can be fast. For Binary and RGB565 images not much can be done.

v923z, 2020-09-21:
OK, then I will insert the pointers, and add the required infrastructure in the functions. Probably conditionally, just for `openmv`. One more thing: we support `(u)int8, (u)int16`, and `float/double`. Do you really need `uint32` in your read function?

kwagyeman, 2020-09-21:
Our internal image types are:

Binary - 1bit per per pixel.
Grayscale - unit8
RGB565

So the results of those methods should be cast to whatever the underlying array data type is. Probably, easiest is just to make it equal to the MP float type. I don’t see folks using this except for float math really. However, the read/write methods should return whatever the data type is. So the return/passed value should be a data type that is equal in size to the largest type you support and then that returned value is cast to the value type the array supports.

v923z, 2020-09-21:
> Our internal image types are:
> 
> Binary - 1bit per per pixel.
> Grayscale - unit8
> RGB565
> 

I overlooked the significance of RGB565: your function can return a single value only, but then it has to support at least three bytes, hence the `uint32`, I guess. In principle, we could return an `rgb = uint8[3]` array, that would be compacter, and it would also simplify the read-out of the values, because you could then access the blue channel by reading rgb[3]. But if it is so, then you need only two dimensions, really, don't you?

kwagyeman, 2020-09-21:
Um, I was thinking you'd call the read/write method per dim. So, an RGB565 image is a 3 channel 2d image. Mmm, but, does ulab support that?

v923z, 2020-09-21:
> Um, I was thinking you'd call the read/write method per dim. So, an RGB565 image is a 3 channel 2d image. Mmm, but, does ulab support that?

I think this is more like a question of what you want to do with your data. In `numpy`, you usually cast your images into a tensor of rank 3, where the third dimension is only of length 3. This is how I traverse the arrays at the moment (for 3 dimensions): 

``` c
#define RUN_SUM(ndarray, type, array, results, rarray, shape, strides, index) do {\
    size_t k = 0;\
    do {\
        size_t l = 0;\
        do {\
            RUN_SUM1((ndarray), type, (array), (results), (rarray), (index));\
            (array) -= (ndarray)->strides[(index)] * (ndarray)->shape[(index)];\
            (array) += (strides)[ULAB_MAX_DIMS - 1];\
            l++;\
        } while(l < (shape)[ULAB_MAX_DIMS - 1]);\
        (array) -= (strides)[ULAB_MAX_DIMS - 2] * (shape)[ULAB_MAX_DIMS-2];\
        (array) += (strides)[ULAB_MAX_DIMS - 3];\
        k++;\
    } while(k < (shape)[ULAB_MAX_DIMS - 2]);\
} while(0)
```

where  `RUN_SUM1((ndarray), type, (array), (results), (rarray), (index));` runs along the innermost (last) axis like so 

``` c
#define RUN_SUM1(ndarray, type, array, results, rarray, index)\
({\
    type sum = 0;\
    for(size_t i=0; i < (ndarray)->shape[(index)]; i++) {\
        sum += *((type *)(array));\
        (array) += (ndarray)->strides[(index)];\
    }\
    memcpy((rarray), &sum, (ndarray)->itemsize);\
    (rarray) += (ndarray)->itemsize;\
})
```

I guess, we would have to change this latter macro so that it can call the function in the pointer

```c
#define RUN_SUM1(ndarray, type, array, results, rarray, index)\
({\
    type sum = 0;\
    for(size_t i=0; i < (ndarray)->shape[(index)]; i++) {\
        sum += (ndarray)->read_array();\
        (array) += (ndarray)->strides[(index)];\
    }\
    memcpy((rarray), &sum, (ndarray)->itemsize);\
    (rarray) += (ndarray)->itemsize;\
})
```

Actually, what you really need is an iterator object, something along these lines: https://github.com/v923z/micropython-ulab/blob/fd7090ce3524c968926762a3ebef6bb8337373b7/code/ndarray.c#L935

Wouldn't this be a more elegant solution than the function pointer?

v923z, 2020-09-21:
Actually, I have to go back on my last comment: the function pointer should probably be a bit more complicated

```c
(uint32_t read_array)(ndarray_obj_t *self, size_t position);
```
because you can't assume that you have to read out the image in a particular direction. Many functions in `numpy` have the `axis` keyword. So, in the example from above, 

```c

#define RUN_SUM1(ndarray, type, array, results, rarray, index, position)\
({\
    type sum = 0;\
    for(size_t i=0; i < (ndarray)->shape[(index)]; i++) {\
        sum += (ndarray)->read_array((position));\
        (position) += (ndarray)->strides[(index)];
    }\
    memcpy((rarray), &sum, (ndarray)->itemsize);\
})

#define RUN_SUM(ndarray, type, array, results, rarray, shape, strides, index, position) do {\
    size_t k = 0;\
    do {\
        size_t l = 0;\
        do {\
            RUN_SUM1((ndarray), type, (array), (results), (rarray), (index), position);\
            (position) -= (ndarray)->strides[(index)] * (ndarray)->shape[(index)];\
            (position) += (strides)[ULAB_MAX_DIMS - 1];\
            l++;\
        } while(l < (shape)[ULAB_MAX_DIMS - 1]);\
        (position) -= (strides)[ULAB_MAX_DIMS - 2] * (shape)[ULAB_MAX_DIMS-2];\
        (position) += (strides)[ULAB_MAX_DIMS - 3];\
        k++;\
    } while(k < (shape)[ULAB_MAX_DIMS - 2]);\
} while(0)
```

kwagyeman, 2020-09-21:
Yeah, so, Ibrahim did this already:

https://github.com/openmv/openmv/blob/master/src/omv/py/py_image.c#L336
https://github.com/openmv/openmv/blob/master/src/omv/py/py_image.c#L283

But, it allocates massive arrays on the heap from our image and it doesn't handle splitting the RGB565 pixels into 3 channels.

I don't know what the best solution looks like. I think you are the expert on this. But, something optimal would be not to allocate huge rows of data. That's fundamentally slower than just calling a function pointer to extract the value at the particular height, width, channel.

v923z, 2020-09-21:
I've got to take off now, but I will think about this, and come back tomorrow.

kwagyeman, 2020-09-21:
As for the type these methods should return, probably the expected return type should be passed. Then they can just return that type format.

E.g. I assume most users will want the image treated as an array of floats so math doesn't overflow. I can modify the .ndarray() the image object can have to you can pass the type of array access should be performed it.

v923z, 2020-09-22:
I think the concept in https://github.com/openmv/openmv/issues/881#issuecomment-696354050 is probably OK, with some refinements. 

1. The position has to be calculated in `ulab`, because that is where you specify, along which dimension you want to iterate. 
2. We should probably just return a `char` pointer, so that the iterator could decide how it wants to interpret the data.
3. The function pointer should just take the position, calculate the pixel value at that position, and fill in the pointer value.

What bugs me a bit is that I haven't yet found out, which `numpy` function we could use to initialise the array. My proposal would be to write a function that simply fills in the members of 

```c
typedef struct _ndarray_obj_t {
    mp_obj_base_t base;
    uint8_t dense;
    uint8_t dtype;
    uint8_t itemsize;
    uint8_t boolean;
    uint8_t ndim;
    size_t len;
    size_t shape[ULAB_MAX_DIMS];
    int32_t strides[ULAB_MAX_DIMS];
    void *array;
    (uint32_t read_array)(ndarray_obj_t *self);
    (void write_array)(ndarray_obj_t *self, uint32_t value);
} ndarray_obj_t;
```
and inserts a pointer to the `data`, and the two functions. There seems to exist no such function in `numpy`. The closest is `ndarray`, which takes a buffer, https://numpy.org/doc/stable/reference/generated/numpy.ndarray.html, but I haven't found a canonical way of supplying functions. I am not against defining our own array creation function, I am only saying that there is no template for that, so we have to iron it out ourselves. 

4. The `read_array`, `write_array` functions, in addition to the position argument, should also take the `dtype`, so that one would know how a string of bytes is to be interpreted. 

5. We might have to add an extra member, `offset`, or something similar, so that you can do backward indexing, like `image[-1:0:-2,:,:]`, or `flip(image)`. In the current implementation, both of these operations just move the `*data` pointer by the required amount, and re-calculate the `strides`. If you want to store images that cannot be indexed, then we can't just move the data pointer in a haphazard way. 

One more comment: this whole concept is going to work for single images only. Some functions could be covered by implementing in-place operators, so, e.g., with that you could calculate the average of 10 images without having to allocate memory for 10 frames, but sorting, or standard deviations could become difficult. I don't know, whether this is a problem.

kwagyeman, 2020-09-22:
Hi, the Image object will just return an ndarray() that's ready to go. You don't need to supply an initializer.

> One more comment: this whole concept is going to work for single images only. Some functions could be covered by implementing in-place operators, so, e.g., with that you could calculate the average [...]

I don't understand how it would handle more than one image...

Oh, I think you are talking about the fact that the data behind the image isn't of the right size to support anything more complex.

Yeah, so, this is a general issue some users had. Some folks wanted to do like *2 on an image of all bytes and they didn't understand that this would just overflow.

...

Question, I linked to our current code above which returns rows of data at a time. Maybe just accessing a row of pixels at time is better? We can very quickly transform a row of data into any target type. This will be faster than single value access. I thought about it and a row isn't so much data to store on the heap.

Like, maybe it's better if there's just an iterator that unlocks a row of values at a time. However, I'm not sure how to handle RGB565 as it's a 3 value tuple in a row.

v923z, 2020-09-23:
> Hi, the Image object will just return an ndarray() that's ready to go. You don't need to supply an initializer.

OK, then I misread your statement. 

> I don't understand how it would handle more than one image...

Let us say you have 10 5-by-5 matrices (images). You could then define an array 
```
a = zeros((5, 5, 10))
for i in range(10):
    a[:,:,i] = matrix[i]
```
and calculate the average matrix as 

```
mean(a, axis=2)
```
But this assumes that you have everything in memory, as a single, contiguous chunk of data. I don't see at the moment, how we can implement this, if you only have the two function pointers, and 10 more or less independent (meaning that their memory locations have nothing to do with each other) images. This is why I said that, for at least the simple cases, we could do something like this

```
a = zeros((5, 5,))
for i in range(10):
    a += matrix[i]
```

but this assumes that the in-place operators are implemented. It's not a big deal, but they are not done yet, and that would cover only those cases, where you can calculate something on the fly. 

> Oh, I think you are talking about the fact that the data behind the image isn't of the right size to support anything more complex.
> 
> Yeah, so, this is a general issue some users had. Some folks wanted to do like *2 on an image of all bytes and they didn't understand that this would just overflow.

Sometimes you really want it to overflow (or, to put it differently, you want to calculate modulo 256), but that is probably not your point. I don't quite know what the use cases are, but if you want your results to have the same dimensions, then you can't really avoid having to allocate that on the heap. 

> Question, I linked to our current code above which returns rows of data at a time. Maybe just accessing a row of pixels at time is better? We can very quickly transform a row of data into any target [...]

That's right, but you might want to iterate along columns. Say, you want to calculate the sum of pixel along an axis, like 

```
a = array.(...)
b = sum(a, axis=1)
```

where then `b` is another array, whose length is equal to the width of `a`. So, in effect, if you can convert columns, too, you would have to move some of the logic from `ulab` into `openmv`. 

I have the feeling that we have run into a serious difficulty here, one for which we won't find a viable template. In `numpy`, things are almost trivial, because the `imread` function just converts the image into the appropriate array, and once you have that, you are in `numpy`-land. I understand now that the finite amount of RAM on a microcontroller poses a severe problem. 

> Like, maybe it's better if there's just an iterator that unlocks a row of values at a time. However, I'm not sure how to handle RGB565 as it's a 3 value tuple in a row.

Right. I don't see, how we could, e.g., convert an RGB565 image into a grayscale one in a single pass.

iabdalkader, 2020-11-18:
@v923z Hello, quick question I'm doing a demo with ulab + audio + our image drawing functions, and I wonder if there's a function to normalize/scale the fft or spectrogram I get with `extras.spectrogram(pcm_samples)` :) ?

v923z, 2020-11-18:
@iabdalkader You can always do something like this: 

```python
import ulab
from ulab import fft, numerical

a = fft.spectrogam(...)
b = a / numerical.max(a)
```
Or do you want to normalise for the area (power)? Then you should probably do 

```python
import ulab
from ulab import fft, numerical

a = fft.spectrogam(...)
b = a / numerical.sum(a)
```
Was it your question?

iabdalkader, 2020-11-18:
@v923z Hi, yes I am doing something very similar to this:

`b = a / numerical.max(a)`

And then I scale the line height using this value like: `line_height = b[0] * image_height` but this leads to the most dominant frequency becoming the max even if it has a magnitude, see what I mean ? so I get spikes like this:

[image]

v923z, 2020-11-18:
> And then I scale the line height using this value like: `line_height = b[0] * image_height` but this leads to the most dominant frequency becoming the max even if it has a small magnitude

I might be missing something. What _should_ happen? Or, perhaps, we can reformulate the question: is there a function in `numpy`, or `scipy` that would fulfil the requirements?

iabdalkader, 2020-11-18:
No it's working fine but I'm asking is there a function to normalize the spectrogram ? Like where I get values that have known lower/upper bounds ? Or how do I find the min and max possible magnitudes in the spectrograms ?

v923z, 2020-11-18:
No, there isn't. We could implement something along those lines, on the other hand, as far as I know, `numpy` doesn't have a dedicated function either (odd), so we would break compatibility there. Also, since the microcontroller resources are scarce, we should probably not add a function just for fun's sake, if the desired effect can be achieved by other means.

kwagyeman, 2020-11-18:
You just take a log. That’s how you make it so one bin isn’t huge.

v923z, 2020-11-18:
> You just take a log. That’s how you make it so one bin isn’t huge.

I don't know, whether @iabdalkader wants to keep it linear.

iabdalkader, 2020-11-18:
I thought I replied to this, anyway yes I will try log to flatten the plot, thanks a lot.

iabdalkader, 2020-11-24:
@v923z Hi, I finally got around to updating MicroPython to 1.13, I'm still working on that right now, but all the OpenMV-specific code is breaking the build (because it's defined in micropython). I can build with 1.13 only if I don't define OPENMV for ulab....Can you please remove all the specific stuff for OpenMV ? I can also send a PR to remove them if you like.

Second questions, when is the next release/tag ? I would like to update, if it's possible to add a tag after removing the openmv stuff that would be great.

One final question, with the current version of ulab if I have a bytes buffer that I want to use to create a short buffer I have to create a short array first then pass that to ndarry, other wise each element is treated as dtype element, for example

```python
import array
import ulab as np

buf = b"\xFF\xFF\xFF\xFF"
print(np.array(buf, dtype=np.uint16))
print(np.array(array.array('H', buf), dtype=np.uint16))
```
This outputs:
```
array([255, 255, 255, 255], dtype=uint16)
array([65535, 65535], dtype=uint16)
```
Is there a way to avoid the double array creation ? Has this issue been fixed ?

v923z, 2020-11-24:
@iabdalkader 

> @v923z Hi, I finally got around to updating MicroPython to 1.13, I'm still working on that right now, but all the OpenMV-specific code is breaking the build (because it's defined in micropython). I [...]

I can do that, no problem. 

> Second questions, when is the next release/tag ? I would like to update, if it's possible to add a tag after removing the openmv stuff that would be great.

If you are willing to wait a couple of days, you would get full `numpy/scipy` compatibility: https://github.com/v923z/micropython-ulab/issues/250. At least, in the sense that if a piece of code works on the microcontroller, then it is going to work on the PC, too. (The single exception is the FFT.) At the moment, the problem is that `numpy` and `scipy` functions are somewhat mixed, so you can't very easily port code. The implementation doesn't require any new code, I simply have to re-arrange bits and pieces. I am almost done, so I hope to roll out version 2.0 at the beginning of next week, perhaps even earlier. 

> One final question, with the current version of ulab if I have a bytes buffer that I want to use to create a short buffer I have to create a short array first then pass that to ndarry, other wise [...]

In the new version, you would probably do this the other way around: you would create the array, and then get the pointer to the data like so: https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=tobytes#tobytes

In that way, you can re-use the buffer as many times as you like. Was this your question?

iabdalkader, 2020-11-24:
> If you are willing to wait a couple of days, you would get full numpy/scipy compatibility: v923z/micropython-ulab#250.
Sure no rush, I can update the submodule later. For now I'll just disable the OPENMV flag so it builds, let me know when it's out.

> In that way, you can re-use the buffer as many times as you like. Was this your question?
Not exactly, I was just trying to save the intermediate array/step and create an int16_t ndarray from bytes in one step.

v923z, 2020-11-25:
> Sure no rush, I can update the submodule later. For now I'll just disable the OPENMV flag so it builds, let me know when it's out.

OK. This shouldn't take long.

> Not exactly, I was just trying to save the intermediate array/step and create an int16_t ndarray from bytes in one step.

Oh, I see. And I also see that there is a problem in `ulab`, when you pass a `bytes` array. I will look into that, thanks for bringing this up!

iabdalkader, 2020-11-25:
@v923z That's true, maybe a bad example, but numpy has frombuffer which works as expected:

```python
In [10]: buf = b'\xFF\xFF\xFF\xFF'
In [11]: np.frombuffer(buf, dtype=np.uint16)
Out[11]: array([65535, 65535], dtype=uint16)
```

v923z, 2020-11-25:
@iabdalkader We could definitely add `frombuffer`.

v923z, 2020-11-25:
@iabdalkader Opened an issue: https://github.com/v923z/micropython-ulab/issues/267

iabdalkader, 2020-11-25:
@v923z That's great thanks!

v923z, 2020-11-26:
@iabdalkader As requested

```python
from ulab import numpy as np

np.set_printoptions(edgeitems=10)
a = bytes(range(0, 16))
print(a)
print()
print(np.frombuffer(a, dtype=np.uint8))
print(np.frombuffer(a, dtype=np.uint16))
print(np.frombuffer(a, dtype=np.float))
```
```python
b'\x00\x01\x02\x03\x04\x05\x06\x07\x08\t\n\x0b\x0c\r\x0e\x0f'

array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15], dtype=uint8)
array([256, 770, 1284, 1798, 2312, 2826, 3340, 3854], dtype=uint16)
array([7.949928895127362e-275, 3.691916204865093e-236], dtype=float64)
```

Could you, please, check out the `numpy-scipy` branch and see if it is what you need: https://github.com/v923z/micropython-ulab/tree/numpy-scipy

I have also removed the `OPENMV` flags. I believe you should be able to compile without having to tamper with the code. In case I am wrong, you should PR patches against that branch.

The only reason for not releasing the code as is is that I still have to fix the documentation. I hope I will have time for that over the weekend. On the other hand, the syntax is `numpy`-compatible now, so you can relatively easily find help. You can also look at https://micropython-ulab.readthedocs.io/en/latest/ulab-intro.html#customising-ulab.

iabdalkader, 2020-11-26:
That's great, thanks a lot! I'll checkout the branch and test my application and get back to you.

iabdalkader, 2020-11-26:
@v923z That branch builds fine, I just had to add the new files and update my example script, however when running it I get a mem-manage fault after a couple of seconds. Unfortunately I can't debug it further right now, seems to be an issue with debug builds and the latest micropython.

v923z, 2020-11-26:
> @v923z That branch builds fine, I just had to add the new files and update my example script, however when running it I get a mem-manage fault after a couple of seconds. Unfortunately I can't debug it further right now, seems to be an issue with debug builds and the latest micropython.

Let me know if you find anything out!

> 
> EDIT: Just to be clear, 0.50.2 is working fine.

https://github.com/v923z/micropython-ulab/tree/0.54.5 is probably better.

iabdalkader, 2020-11-26:
@v923z Was able to build with debugging so I may be able to narrow it down for you, and it happens if I use any function, mean, abs, frombuffer.

iabdalkader, 2020-11-26:
@v923z Hi, sorry it was a false alarm, it's actually a gc-collect bug has nothing to do with ulab :man_facepalming:

iabdalkader, 2020-11-26:
@v923z There's a different issue though, does `frombuffer` make a copy or return a reference to the buffer ?

[36 more replies]

closed 2024-01-14 by kwagyeman