openmv/openmv #888: find_rects() printout for overlapping rectangles

timhenderson17, 2020-08-25:
### Steps to reproduce
1. Use find_rects() function (with threshold 8000 but have had issue with multiple thresholds)
2. Point camera at an area with many rectangles

### Expected behaviour
Nothing should be printed

### Actual behaviour
When (I'm assumig) overlapping rectangles are detected it prints: "uh oh, no preference for overlappingdetection"

^The space is missing between overlapping and detection in the message

### Describe your setup
Using the H7, with OV7725, RGB565, QQVGA

**IDE version:**
2.5.1

**Firmware version:**
3.6.4

**Operating system:**
Windows

kwagyeman, 2020-08-25:
From here: https://github.com/openmv/openmv/blob/master/src/omv/img/apriltag.c#L11884

kwagyeman, 2021-01-03:
https://github.com/openmv/openmv/pull/1095

closed 2021-01-03 by kwagyeman