openmv/openmv #889: Questions about standby, sleep and stop power modes

acutetech, 2020-08-27:
IDE low power examples use machine.deepsleep() and machine.sleep(). From what I can see, these correspond to STM32 processor chip Standby and Sleep states, respectively. Is that correct?

Standby state is understandable as it gives the lowest possible processor power consumption (c. 2.4uA according to the data sheet); however resuming from standby forces a processor reset.

Sleep state still gives a high power consumption - in the region of 50-130mA according to the data sheet, depending on several variables. This is verified in my tests. This not a very impressive or useful saving if you are trying to run from batteries. Execution resumes from the point the processor enters sleep.

Have you considered or tested the processor stop state? This seems to give a current in the range 1.3 - 0.2mA which is a much more useful low-power current. From what I can see, execution will resume from where the processor entered stop mode. Both the RTC and GPIO pin interrupts can cause a resume, AFAIK. I would imagine that for most users the experience would be the same as for machine.sleep() but with MUCH lower power consumption.

From what I can see, table 34 of the RM0433 Reference Manual summarises low power states. (I know it looks complex... )

kwagyeman, 2020-08-27:
As OpenMV is a two person company running in our free time... we really haven't looked that carefully. We'd appreciate code PRs and help digging into this.

acutetech, 2020-09-01:
I have done some initial digging into the code and report my interim findings here, for the record. (There is much I don't understand about the inner workings of the Python-STM32 mapping.)

The ST HAL layer defines three power-down modes (SLEEP, STOP, STANDBY) which are described here:
https://github.com/openmv/openmv/blob/d6b5d3370522e23796a2b67ebda68dda0f6dcd0b/src/sthal/h7/src/stm32h7xx_hal_pwr.c#L162

*** SLEEP ***
HAL_PWR_EnterSLEEPMode() is available in openmv-master\src\sthal\h7\src\stm32h7xx_hal_pwr.c but apparently not used by Python.
    https://github.com/openmv/openmv/blob/d6b5d3370522e23796a2b67ebda68dda0f6dcd0b/src/sthal/h7/src/stm32h7xx_hal_pwr.c#L377
    
*** STOP ***
I suspect that pyb.stop() and machine.sleep() and machine.lightsleep() all initiate a STOP:

Looks like there is a mapping of MP_QSTR_stop to machine_lightsleep_obj in ports/stm32/modpyb.c
    https://github.com/openmv/micropython/blob/1e4e3ec8053c01b52596521e946a655540a5aa68/ports/stm32/modpyb.c#L153
_and_ MP_QSTR_sleep _and_ MP_QSTR_lightsleep to machine_lightsleep_obj in /ports/stm32/modmachine.c
    https://github.com/openmv/micropython/blob/1e4e3ec8053c01b52596521e946a655540a5aa68/ports/stm32/modmachine.c#L375 
      
which presumably leads to routine machine_lightsleep() in modmachine.c:
    https://github.com/openmv/micropython/blob/1e4e3ec8053c01b52596521e946a655540a5aa68/ports/stm32/modmachine.c#L338
which calls powerctrl_enter_stop_mode() in powerctrl.c
    https://github.com/openmv/micropython/blob/31ba06ce845938e4fff9a34163311a07d0e22cab/ports/stm32/powerctrl.c#L333 
which in turn calls HAL_PWR_EnterSTOPMode() in  openmv-master\src\sthal\h7\src\stm32h7xx_hal_pwr.c
    https://github.com/openmv/openmv/blob/d6b5d3370522e23796a2b67ebda68dda0f6dcd0b/src/sthal/h7/src/stm32h7xx_hal_pwr.c#L419

*** STANDBY ***
I suspect that pyb.standby() and machine.deepsleep() both initiate a STANDBY

Looks like there is a mapping of MP_QSTR_standby to machine_deepsleep_obj in ports/stm32/modpyb.c
    https://github.com/openmv/micropython/blob/1e4e3ec8053c01b52596521e946a655540a5aa68/ports/stm32/modpyb.c#L154
and MP_QSTR_deepsleep to machine_deepsleep_obj in /ports/stm32/modmachine.c
    https://github.com/openmv/micropython/blob/1e4e3ec8053c01b52596521e946a655540a5aa68/ports/stm32/modmachine.c#L377 
       
which presumably leads to routine  machine_deepsleep()  in modmachine.c: 
    https://github.com/openmv/micropython/blob/1e4e3ec8053c01b52596521e946a655540a5aa68/ports/stm32/modmachine.c#L348
which calls powerctrl_enter_standby_mode()  in powerctrl.c
    https://github.com/openmv/micropython/blob/31ba06ce845938e4fff9a34163311a07d0e22cab/ports/stm32/powerctrl.c#L435
which in turn calls HAL_PWR_EnterSTANDBYMode() in  openmv-master\src\sthal\h7\src\stm32h7xx_hal_pwr.c
    https://github.com/openmv/openmv/blob/d6b5d3370522e23796a2b67ebda68dda0f6dcd0b/src/sthal/h7/src/stm32h7xx_hal_pwr.c#L482 

So unlike my initial assumption in my first post, it does look like the Sleep mode is in fact used, not the Stop mode that I suspected. So the big unanswered question is why the Sleep current is dropping to c. 50mA and not to c. 1mA as the data sheet suggests....

kwagyeman, 2021-01-29:
We got the current down to 10mA. Not sure how to get it lower. It seems to be coming from the processor.

It's probably because clocks are running fast and still on. We are currently really improving driver support every day.

acutetech, 2021-02-09:
I saw the post above re "10mA", then soon after news of release 2.6.9. I was not sure whether the "10mA" post implied you had made changes that would be available in 2.6.9, but having tried 2.6.9 it seems that there are changes. I found the following (setup is H7, running from SD card, 4.5V battery, external interrupt pin used to wake the board):

Before (3.8.0 firmware): machine.sleep() current 30mA, reduces to 25mA if clock frequency dropped from 400MHz to 50MHz, reduced to 7mA if the sensor and the sensor code is removed. Therefore, most current is associated with the sensor.

After (3.9.2 firmware): machine.sleep() current 7.1mA with or without the sensor. Drops to 6.9mA when CPU frequency is reduced from 400MHz to 50MHz. (With a second borad the available frequencies  are 60MHz and 480MHz and the currents are 8.2mA and 8.6mA respectively.)

So a great improvement - thanks!

I'd be interested to know what changes you made.

Is there any way of going back to the earlier 3.8.0 firmware so I can repeat these tests?

iabdalkader, 2021-02-09:
Yes I made some fixes to low power modes, mainly making sure QSPI and SDRAM switch to low power on entry to low power modes, and disable DCMI pins (they seem to have been leaking current when the sensor is shutdown). You can find changes in the commit history and yes you can downgrade any older firmware from the github releases.