openmv/openmv #891: ExtInt on P6 fails to generate an interrupt in some cases

acutetech, 2020-08-31:
I have written two examples that look out for interrupts from GPIO pins. The initial (more complex) example works with all pins except P6 (which is the only pin on Port A). I then wrote a simplified version, intending to post it here with  a bug report. However, the simple example works fine on all pins, including P6. AFAIK the essential logic of both examples is the same. Very weird.  Ideas?

Simple example (works):

```
# External Interrupt example

import pyb, utime
from pyb import Pin, ExtInt

global event
global irqPin
event = False
irqPin = -1

# Don't do much in this - it is an ISR
def switchEvent(x):
    global event
    global irqPin
    event = True
    irqPin = x

# Select which pin P0-P9 to use. Short this pin to GND through a switch for an interrupt
pin = Pin("P6", Pin.IN, Pin.PULL_UP)
ext = ExtInt(pin, ExtInt.IRQ_FALLING, Pin.PULL_UP, switchEvent)

print("ExtInt test")

while(True):
    if (event):
        print("Boo! " + str(irqPin))
        # Now wait at least a second, and until the pin goes high again
        utime.sleep_ms(1000)
        while(pin.value() == 0):
            utime.sleep_ms(20)
        event = False   # Don't clear the flag until the end, as it might have been set again.
```

More complex example (works on all except P6):

```
# Take snapshot on GPIO pin interrupt

import sensor, image, pyb, utime, uos
from pyb import Pin, ExtInt

global event
global counter
event = False
counter = 0

print("*** Use GPIO to make a snapshot and save it to disk ***")

# Don't do much in this - it is an ISR
def switchEvent(x):
    global event
    global counter
    event = True
    counter = counter + 1

# Delete the old files
def deleteOldFiles():
    directory = "pics"

    # First see if the pics directory exists
    dirExists = False
    for filename in uos.listdir():
        if (filename == directory):
            dirExists = True

    if (dirExists):
        # Delete all of our picture files
        i = 0
        for filename in uos.listdir(directory):
            if filename.endswith(".jpg") and filename.startswith("p_"):
                i = i + 1
                uos.remove(directory + "/" + filename)
            else:
                continue

        print("Erased " + str(i) + " old pictures")

    else:
        print("Creating '" + directory + "' directory")
        uos.mkdir(directory)

# Define the GPIO pin to use: should be able to select P0 - P9
activeState = 0 # This is the active low option - switch to GND
pin = Pin("P6", Pin.IN, Pin.PULL_UP)
ext = ExtInt(pin, ExtInt.IRQ_FALLING, Pin.PULL_UP, switchEvent)

#activeState = 1 # This is the active high option - switch to 3V3
#pin = Pin("P6", Pin.IN, Pin.PULL_DOWN)
#ext = ExtInt(pin, ExtInt.IRQ_RISING, Pin.PULL_DOWN, switchEvent)

RED_LED_PIN = 1
GREEN_LED_PIN = 2
BLUE_LED_PIN = 3

deleteOldFiles()

sensor.reset() # Initialize the camera sensor.
sensor.set_pixformat(sensor.RGB565) # or sensor.GRAYSCALE
sensor.set_framesize(sensor.VGA) # or sensor.QQVGA (or others)
sensor.skip_frames(time = 1000) # Let new settings take affect.

print("Ready: pin is now " + str(pin.value()))
sTime = utime.time()

while(True):
    if (event):
        sensor.skip_frames(time = 200)

        pTime = utime.time() - sTime
        fileName = "pics/p_" + str(pTime) + ".jpg"
        print("Saving as " + fileName + " Counter=" + str(counter))

        pyb.LED(RED_LED_PIN).on()
        pyb.LED(GREEN_LED_PIN).on()
        pyb.LED(BLUE_LED_PIN).on()

        sensor.snapshot().save(fileName)

        pyb.LED(RED_LED_PIN).off()
        pyb.LED(GREEN_LED_PIN).off()
        pyb.LED(BLUE_LED_PIN).off()

        # Now wait at least a second, and until the pin goes inactive again
        utime.sleep_ms(1000)
        while(pin.value() == activeState):
            utime.sleep_ms(20)
        # Don't clear the flag till the end, as interrupts may have been happening in the background
        event = False
        print("-")

```

kwagyeman, 2020-08-31:
P6 is the ADC DAC pin. It might have special behavior.

iabdalkader, 2020-10-16:
EXTI9-5 is shared with the sensor VSYNC IRQ and it gets disabled in sensor.reset()... If you move the extint code after reset it will re-enable the IRQn.

```
sensor.reset()
.... etc...
ext = ExtInt(pin, ExtInt.IRQ_FALLING, Pin.PULL_UP, switchEvent)
....
some other code
```

closed 2020-10-16 by iabdalkader