openmv/openmv #920: Issue in April Tag example code

nitanshnagpal1, 2020-10-09:
### Steps to reproduce
1. Run the april tag example code.
2. In serial terminal it print random april tag code even when we do not show april tag.

### Expected behaviour
Tell us what should happen
It prints april tag code even when we do not show april tag to the camera.

### Actual behaviour
Tell us what happens instead

It should't print the april tag code when we do not show the april tag to camera

### Describe your setup

**IDE version:** 2.6.5

**Firmware version:**

**Operating system:** Windows 10

### For developers

**GCC version:**

**Build target:**

kwagyeman, 2020-10-09:
What tag family are you using? For 36H11 it’s very hard to get a false positive, however, this can happen. It’s not a bug. If you are using a tag family with a lower hamming code it can happen more often.

nitanshnagpal1, 2020-10-10:
Thankyou for your response. Actually it was showing false positive for TAG25H9. We will not use this tag in our application.

closed 2020-10-10 by nitanshnagpal1

kwagyeman, 2020-10-10:
Yeah, that will have more false positives.