openmv/openmv #924: find_rects issue with roi argument

kidswong999, 2020-10-11:
find_rects will crash with roi argument, I think firmware has bug.
firmware: 3.6.8
board: OpenMV 4 H7
code:
```
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565) # grayscale is faster (160x120 max on OpenMV-M7)
sensor.set_framesize(sensor.QQVGA)
sensor.skip_frames(time = 2000)
clock = time.clock()

while(True):
    clock.tick()
    img = sensor.snapshot()

    # `threshold` below should be set to a high enough value to filter out noise
    # rectangles detected in the image which have low edge magnitudes. Rectangles
    # have larger edge magnitudes the larger and more contrasty they are...

    for r in img.find_rects(roi = (65,45,30,30), threshold = 10000):
        img.draw_rectangle(r.rect(), color = (255, 0, 0))
        for p in r.corners(): img.draw_circle(p[0], p[1], 5, color = (0, 255, 0))
        print(r)

    print("FPS %f" % clock.fps())
```

Thanks

kwagyeman, 2020-10-11:
Will fix. The scaling code updates will fix this along with dealing with other issues.

kwagyeman, 2020-10-20:
Found the issue. ROI was not being used in all places.

closed 2020-10-20 by kwagyeman