openmv/openmv #933: Can't save VGA RGB image as BMP?

acutetech, 2020-10-24:
When I try to save a RGB565 VGA image to disk I get "OSError: Image is not BMP!"

Is this a feature or a bug?

It can easily be reproduced by taking the snapshot_on_movement.py example and changing the sensor.QVGA to sensor.VGA. The problem does not arise with GREYSCALE VGA images.

I am guessing this is a memory constraint? Perhaps it could be documented (or perhpas it is?), or perhaps the error message could be clearer.

I was attempting to save at the highest possible quality, which I guessed would have been RGB rather than greyscale, assuming some information would have been lost in B&W (perhaps this is not so?). Is there a way of saving a full-scale RGB image as .BMP?

I am using  H7 camera, Firmware 3.6.8, IDE 2.6.5

kwagyeman, 2020-10-24:
Its because the RGB565 image is actually a bayer image. We silently put the camera in Bayer mode when you ask for RGB565 but we don't have enough RAM. Then, what you are seeing is the processor debayering the image and then jpeg compressing it.

JPG has the debayer front end. So, you should be able to save images there. I can add this to BMP. It's been on my todo list for a while to support read-only functions for BAYER images (saving a file is read-only access to the bayer image as I don't have to modify it to save it to disk).

acutetech, 2020-10-25:
Don't do the work for my sake - JPG works and greyscale works. I reported the issue in the spirit of flagging unexpected issues, in case it was helpfu to you or others.

kwagyeman, 2020-10-25:
My goal in the next year is to fix all this I/O stuff. With Arduino getting behind us this needs to be fixed.