openmv/openmv #991: Error building openmv from source for OpenMV CamH7 Plus

gardnmc, 2020-11-22:
Great project guys.

I've followed the instructions to set up the build from source and can successfully build for the H7, for example...

make v=1 TARGET=OPENMV4
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Warning: FROZEN_MPY_DIR is deprecated in favour of FROZEN_MANIFEST
   text	   data	    bss	    dec	    hex	filename
  19984	    224	  10692	  30900	   78b4	/home/mike/repos/openmv/src/build/bin/bootloader.elf
   text	   data	    bss	    dec	    hex	filename
1820104	   2312	 848836	2671252	 28c294	/home/mike/repos/openmv/src/build/bin/firmware.elf

However, when building for the H7 Plus I get some pin mapping errors.
Using Ubuntu 16.04 LTS
gcc-arm-none-eabi-7-2018q2
Any pointers appreciate
Build output below, let me know if I need to provide any further info.  Thanks.

make v=1 TARGET=OPENMV4P
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Warning: FROZEN_MPY_DIR is deprecated in favour of FROZEN_MANIFEST
CC boards/OPENMV4P/bdev.c
In file included from ./mpconfigport.h:31:0,
                 from /home/mike/repos/openmv/src/micropython/py/mpconfig.h:62,
                 from /home/mike/repos/openmv/src/micropython/py/mphal.h:29,
                 from /home/mike/repos/openmv/src/micropython/drivers/bus/qspi.h:29,
                 from ./qspi.h:29,
                 from boards/OPENMV4P/bdev.c:1:
/home/mike/repos/openmv/src/micropython/ports/stm32/boards/OPENMV4P/mpconfigboard.h:93:42: error: 'pin_G6' undeclared here (not in a function); did you mean 'pin_A0'?
 #define MICROPY_HW_QSPIFLASH_CS         (pin_G6)
                                          ^
boards/OPENMV4P/bdev.c:7:12: note: in expansion of macro 'MICROPY_HW_QSPIFLASH_CS'
     .cs  = MICROPY_HW_QSPIFLASH_CS,
            ^~~~~~~~~~~~~~~~~~~~~~~
/home/mike/repos/openmv/src/micropython/ports/stm32/boards/OPENMV4P/mpconfigboard.h:94:42: error: 'pin_F10' undeclared here (not in a function); did you mean 'pin_A10'?
 #define MICROPY_HW_QSPIFLASH_SCK        (pin_F10)
                                          ^
boards/OPENMV4P/bdev.c:8:12: note: in expansion of macro 'MICROPY_HW_QSPIFLASH_SCK'
     .clk = MICROPY_HW_QSPIFLASH_SCK,
            ^~~~~~~~~~~~~~~~~~~~~~~~
/home/mike/repos/openmv/src/micropython/ports/stm32/boards/OPENMV4P/mpconfigboard.h:95:42: error: 'pin_F8' undeclared here (not in a function); did you mean 'pin_C8'?
 #define MICROPY_HW_QSPIFLASH_IO0        (pin_F8)
                                          ^
boards/OPENMV4P/bdev.c:9:12: note: in expansion of macro 'MICROPY_HW_QSPIFLASH_IO0'
     .io0 = MICROPY_HW_QSPIFLASH_IO0,
            ^~~~~~~~~~~~~~~~~~~~~~~~
/home/mike/repos/openmv/src/micropython/ports/stm32/boards/OPENMV4P/mpconfigboard.h:96:42: error: 'pin_F9' undeclared here (not in a function); did you mean 'pin_F8'?
 #define MICROPY_HW_QSPIFLASH_IO1        (pin_F9)
                                          ^
boards/OPENMV4P/bdev.c:10:12: note: in expansion of macro 'MICROPY_HW_QSPIFLASH_IO1'
     .io1 = MICROPY_HW_QSPIFLASH_IO1,
            ^~~~~~~~~~~~~~~~~~~~~~~~
/home/mike/repos/openmv/src/micropython/ports/stm32/boards/OPENMV4P/mpconfigboard.h:97:42: error: 'pin_F7' undeclared here (not in a function); did you mean 'pin_F9'?
 #define MICROPY_HW_QSPIFLASH_IO2        (pin_F7)
                                          ^
boards/OPENMV4P/bdev.c:11:12: note: in expansion of macro 'MICROPY_HW_QSPIFLASH_IO2'
     .io2 = MICROPY_HW_QSPIFLASH_IO2,
            ^~~~~~~~~~~~~~~~~~~~~~~~
/home/mike/repos/openmv/src/micropython/ports/stm32/boards/OPENMV4P/mpconfigboard.h:98:42: error: 'pin_F6' undeclared here (not in a function); did you mean 'pin_F7'?
 #define MICROPY_HW_QSPIFLASH_IO3        (pin_F6)
                                          ^
boards/OPENMV4P/bdev.c:12:12: note: in expansion of macro 'MICROPY_HW_QSPIFLASH_IO3'
     .io3 = MICROPY_HW_QSPIFLASH_IO3,
            ^~~~~~~~~~~~~~~~~~~~~~~~
../../py/mkrules.mk:47: recipe for target '/home/mike/repos/openmv/src/build/micropython/boards/OPENMV4P/bdev.o' failed
make[1]: *** [/home/mike/repos/openmv/src/build/micropython/boards/OPENMV4P/bdev.o] Error 1
Makefile:643: recipe for target 'FIRMWARE_OBJS' failed
make: *** [FIRMWARE_OBJS] Error 2

kwagyeman, 2020-11-22:
git submodule update --init

iabdalkader, 2020-11-22:
Seems your submodules are not updated, try: `git submodule update --init --recursive`

closed 2020-11-22 by iabdalkader

gardnmc, 2020-11-22:
Tried your suggestion - exactly the same results.  So double checked by repeating the installation with a new fresh install using the  instructions at  https://github.com/openmv/openmv/blob/master/src/README.md
Again, same result, so followed by repeating git submodule update --init --recursive
Exactly the same result.  Are the instructions on github up to date, or is there a new procedure somewhere?
Thanks.

iabdalkader, 2020-11-22:
Yes these build instructions are up to date, if you clone the repo recursively you don't need any more git commands:

```git clone --recursive https://github.com/openmv/openmv.git```

Then you have to build mpy-cross (only the first time):
```make -j8 -C src/micropython/mpy-cross```
 
You can see here in Travis's log the build is working fine so it must be your machine:

https://travis-ci.com/github/openmv/openmv/pull_requests

Our Travis build script is here:
https://github.com/openmv/openmv/blob/master/.travis.yml#L21

gardnmc, 2020-11-23:
Yes - that's exactly what I'm doing, and it matches  
And oddly, the problem only exists for the make v=1 TARGET=OPENMV4P, everything else works fine.
I'll try building a new VM