openmv/openmv #994: sensor.snapshot() fails on portenta h7

idohalamit, 2020-11-24:
### Steps to reproduce
1. Connect Portenta h7 with vision shield to IDE
2. Run any micropython code using sensor module that actually interacts with the camera sensor, such as:
    - sensor.skip_frames()
    - sensor.snapshot()

### Expected behaviour
The camera will return a snapshot, skip frames or whatever according to the used function
### Actual behaviour
The script throws  `RuntimeError: Capture Failed: -4`

### Describe your setup

**IDE version:** 2.6.7

**Firmware version:** Tried on 3.6.8, 3.6.9, 3.7.0

**Operating system:** Windows 10 1909

### For developers

**Build target:** Arduino Portenta H7 with vision shield.

[image]

kwagyeman, 2020-11-24:
Hi, I have this same issue. Unfortunately, your camera is likely dead. Please check the back of your camera and see if there are burn marks on it or melted plastic.

Arduino literally put a metal camera back on top of metal traces behind the camera. Mine shorted and broke. In particular, my camera has some scorch marks on it and melted plastic on the back of it.

kwagyeman, 2020-11-24:
Pointing Arduino folks to this thread.

idohalamit, 2020-11-24:
Camera is literally brand new. No marks of deformation either.

kwagyeman, 2020-11-24:
Mmm, okay, it could be something else. @iabdalkader your camera still works right with v3.7.0? Any thoughts?

iabdalkader, 2020-11-24:
Hi, thanks for the detailed bug report! I can confirm my Portenta with firmware 3.7.0 is working just fine (Note you shouldn't use any of the older firmwares). I'm not sure why yours isn't working, you should double check the board to board connectors and the sensor ribbon connector for issues, also make sure the sensor back isn't shorting as Kwabena mentioned.
Note I have a development camera board it's slightly different than the production boards, we'll have to wait for someone from Arduino to confirm if the firmware is working with production boards. 
[image]

sebromero, 2020-12-02:
@idohalamit May sound silly, but please check if the high density connector has fully snapped in. Press the shield firmly onto the board. I've seen other people having this issue before.

idohalamit, 2020-12-04:
Oh my god I am so dumb. I even went to the trouble of screwing them together before I pressed them strongly enough.

closed 2020-12-04 by idohalamit

sebromero, 2020-12-04:
It happens, don't worry. I'm glad we resolved it.