openmv/openmv #1027: Some potential risk in the py_tf.c

CristXu, 2020-12-10:
hi，
       When i check the code of the py_tf.c:220, the code set the fscale = 1.0f / 255.0f; and when the type of the input tensor is float, 
we use this var to process the pixels, like:
[image]
        After this action, the pixel will be the one between (0 - 1) , and in float type, but this works when the model is also accepted 
inputs in this range. 
        but if we pass a model trained by the datas which are bewteen(-1, 1), such as  we processed the data before thain, by (x / 128.0 - 1). this will not work, even though the type of input is float.( the model is strange that the tflite converter will leave the first node as float , and other as int8). 
       So,  I want to check if any other code do this work? or the code can only accept / inference a tflite model with a input data between (0, 1).

Regards.

RockySong, 2020-12-10:
For image data, we usually get uint8 input, and map [0, 255] -> [-1, 1) for training. For quantized model, just need to map [0, 255] to [-128, 127].
I see the above code also done this with "^ shift" (there is a shift = signed_or_unsigned ? 128 : 0;).
So my understanding is if tf tool can generate a model with u8 input and treat as s8, the whole s8 CMSIS-NN operators can be used.
But it seems tf tool does not generate u8 input model.

kwagyeman, 2020-12-10:
Our front end is designed for Edge Impulse which trains the network using 0-1 float ranges.

If possible, just don't use negative values. Also, from a mapping standpoint this is also less accurate. 0-1 is preferred. Given Edge Impulse is our deployment system and the above approach works for old mobilenet networks and etc. I would argue to please not use -1/+1 floats.

kwagyeman, 2020-12-10:
The CMSIS-NN code needs -128/127, so, for optimal performance the network should use values in this range internally.

kwagyeman, 2020-12-10:
TensorFlow for some reason likes to leave the input layer as floats while the internal layers and quantized. It takes more effort to force all layers to int8_t.

CristXu, 2020-12-11:
> The CMSIS-NN code needs -128/127, so, for optimal performance the network should use values in this range internally.

Hi, thank you for yor reply, i have another question, the mobilnet inside the openmv ide is quanttized with U8, so it will not be optimizeized, or in other words, it will use tflite's reference api when do the inference that will effect the performance. Is this a risk? 

Regards

kwagyeman, 2020-12-11:
No one use that mobilenet. It's just a demo. It's quite useless. As for the person detector, that is more useful. However, we have to use the uin8_t version versus the int8_t version because of the network size. If you have more flash space you can swap to the faster person detector network.

kwagyeman, 2020-12-11:
Again, Edge Impulse TF Lite models are int8_t which are what folks use.

CristXu, 2020-12-11:
> Again, Edge Impulse TF Lite models are int8_t which are what folks use.

Ok， Thanks

CristXu, 2020-12-15:
Hi，when run the py_tf_classify, i found that the code will load the model once we call the tf.classify(), if the model is a little big, this will costs  too much time, and also will effect the inference time.
So, what's your opinion?  Or if there are any other consideration?
[image]

kwagyeman, 2020-12-15:
You can load the model onto the frame buffer first and the re-use the loaded model.

https://docs.openmv.io/library/omv.tf.html#tf.tf.load

Then you run classify on the model object itself.

CristXu, 2020-12-16:
Thanks for your reply, but seems that the example inside the openmv ide, does not use this way : call the tf.load first then the tf.classify.  it uses the tf.classify only, So do you have any plan to update this example(tf_mobilenet_search_whole_window.py) ? otherwise it is maybe misleading the users.

kwagyeman, 2020-12-16:
I don’t use load in the examples because it uses up frame buffer memory. It’s more flexible not to use load. Users who care about performance can reread the API.

CristXu, 2020-12-16:
Ok, thank you very much.

kwagyeman, 2020-12-16:
Thanks, also, I apologize for my tone in the above comment. Was just trying to be brief when writing the response from my phone.

iabdalkader, 2020-12-16:
Hello :wave: please don't forget to close the issue if the issue was resolved.

CristXu, 2020-12-16:
That's alright. Solved the problem is the important thing.

> Thanks, also, I apologize for my tone in the above comment. Was just trying to be brief when writing the response from my phone.

That's alright. I didn’t detect any  with your tone. Thank you again for your reply, i will close this issue soon.

closed 2020-12-16 by CristXu