openmv/openmv #1032: RTC.wakeup() shows unreliable accessing of global variables

acutetech, 2020-12-16:
RTC.wakeup() is designed to generate periodic events. I would expect the wakeup() callback to be able to get and set global variables reliably, but this appears not to be the case. Furthermore, actual behaviour varies depending on whether the callback routine parameter in the wakeup() call includes or does not exclude the dummy variable, and whether wakeup() is invoked once only, or every time the timer event occurs.

To see the issue, run the code below. Uncomment one of the four invocations of wakeup() at lines 10, 11, 14 or 15 and observe the different behaviours.

### Expected behaviour
(a) In all 4 cases, the red LED should toggle every 1s, since "triggered" should be set true at line 8 in the callback, and should be read by the if statement at line 18.
(b) It should not be necessary to call wakeup() every time the timer event occurs (at lines 14 or 15).
(c) The green LED should not toggle. If it does it indicates a bug: for some reason the value of "triggered" that should be changed at line 8 is not read by the if statement at line 18.

### Actual behaviour
line 10 used - green LED toggles every 1s
line 11 used - red LED toggles once and stays on. Then the green LED toggles every 1s 
line 14 used - green LED toggles every 1s
line 15 used - red LED toggles every 1s (the only "correct" case)

Note it is not sufficient to say "OK then use the line 15 implementation" since other examples show this does not always work, and that in some circumstances the callback cannot get the value of global variables. See https://forums.openmv.io/viewtopic.php?f=6&t=2200 and https://forums.openmv.io/viewtopic.php?f=6&t=2201 for longer examples and stranger behaviour. The second of these examples behaves as though the callback changes the value of a global every second time.

Looks like some mess up with pointers or stacks or ISR handling or something. Or perhaps the main thread resues execution before the callback code is executed.

### Describe your setup
 I used an OpenMV H7 with IDE 2.6.8 and firmware 3.8.0. I run the code from the IDE using local flash (not an SD card).

`
import machine, pyb

global triggered
triggered = False

def test(arg):
    global triggered
    triggered = True

#pyb.RTC().wakeup(1000, test)
pyb.RTC().wakeup(1000, test(1))

while (True):
    #pyb.RTC().wakeup(1000, test)
    #pyb.RTC().wakeup(1000, test(1))
    machine.sleep() # Wait for the 

    if (triggered):
        pyb.LED(1).toggle() # toggle red LED
    else:
        pyb.LED(2).toggle() # toggle green LED

    triggered = False
`

acutetech, 2020-12-16:
I don't know why the "insert code" did not work. I will try to attach a file instead. This is a .py file but I had to rename it as .txt 
[rtc_wakeup_2.txt](https://github.com/openmv/openmv/files/5701434/rtc_wakeup_2.txt)

iabdalkader, 2020-12-16:
The RTC callback is not called immediately it just gets scheduled to be called back later, if you wait long enough triggered will be set eventually. To prove this just add a single sleep line after machine.sleep() which allows the scheduler to run:

```
machine.sleep()
time.sleep(0.1)  <---- callback gets called
```

I can't fix that, I don't see any API to set the RTC IRQ as a hard IRQ, you should raise this issue with micropython and feel free to mention me if there's an update/fix I'll patch our code.

> pyb.RTC().wakeup(1000, test(1))

This is very wrong, it just passes None as the callback, it probably works because it calls test which sets triggered (explains why it works in a loop).

acutetech, 2020-12-16:
Thanks Ibrahim - your suggestion of adding a delay fixed all the instances of problems. I know this is not a problem of your making, but perhaps you could add a note in your own documentation of wakeup().

w.r.t. the C code - I did not read it all, but:
(1) This code seems to be the handler for external interrupt pins. Does it also handle the RTC interrupt as well?
(2) If so, then am I right that pin interrupts are handled immediately and only the RTC interrupt is scheduled for later?
(3) If so then that also explains why my pin interrupts did not show a problem, only the RTC interrupt.

iabdalkader, 2020-12-16:
Yes to all of the above, Extint module sets IRQ hard = true by default.

acutetech, 2020-12-16:
Experimentally I find I can reduce the delay to 1ms, using the time.sleep_ms() or utime.sleep_ms() calls. 

Bizarrely, as I tried to refine the delay time further, I used sleep_us() and find that utime.sleep_ms(1) is not the same as utime.sleep_us(1000). Using utime.sleep_us(1000) or even utime.sleep_us(200000) results in green LED flashing, as though no delay is being added.... I don't propose to investigate this, though I think it has to be a bug rather than a feature.

I attach a refactored example that appears reliable to me, with some comments. You might want to use this in your own examples.

`
import machine, pyb, time

global triggered
triggered = False

def callback(arg):
    global triggered
    triggered = True

pyb.RTC().wakeup(1000, callback)

while (True):

    machine.sleep()
    # At this point the RTC will have woken us but the callback may not have been executed.
    # This delay is essential to allow the callback to execute and update 'triggered'
    time.sleep_ms(1)    # Required minimum delay is not known. Do not use sleep_us()

    if (triggered):
        # Toggle green LED to indicate correct behaviour.
        # 'triggered' is seen as having changed
        pyb.LED(2).toggle()
    else:
        # Toggle red LED to indicate incorrect behaviour.
        # 'triggered' is NOT seen as having changed
        pyb.LED(1).toggle()

    triggered = False
`

iabdalkader, 2020-12-17:
It seems `sleep_ms` causes the scheduler to run, but not `sleep_us`... We don't have any RTC examples with callback, but I'm watching the issue in micropython if it's a bug and it gets fixed I'll fix it, otherwise will add a note in the docs.

closed 2020-12-17 by iabdalkader