openmv/openmv #1071: Receive Extended CANBus Frame on H7/H7+

roboticsgeek, 2020-12-30:
**Describe the bug**
I mentor on a FIRST Robotics Challenge (FRC) team that is using OpenMV cameras and interfacing them to the CANBus on our system for robustness. FRC requires the use of a FRC standard extended addressing scheme on the single allowed CANBus in their rules. We would like to be able to send CANBus messages to/from new H7/H7+ cameras like we could do with previous M7 cameras.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect OpenMV M7 to OpenMV H7 /w CANBus.
2. Use the can.py example program to send normal 11-bit frames to H7.
3. Change can.py to use extended frames and pick an extended ID to send to H7.
4. H7 never reads anything from the CANBus, eventual timeout.

**Expected behavior**
Receive either types of CANBus frames on H7 / H7 Plus depending on CANBus initialization.

**Please complete the following information**
 - IDE version: 2.6.8
 - Firmware version: MicroPython: v1.13-r22 OpenMV: v3.8.0 HAL: v1.9.0 BOARD: OPENMV4P-STM32H743
 - Board: OPENMV-H7+/OPENMV-H7
 - OS: [e.g. Windows/MacOS]

kwagyeman, 2021-01-31:
When do you need this by? I might be able to do this next month.

roboticsgeek, 2021-02-01:
That would be great! We would take it whenever we can get it. At the moment we're coping with 1-way protocol only.

iabdalkader, 2022-04-02:
I know this is terribly late, but I've been working on improving CAN FD support in Micropython, and the following has been implemented/fixed:

- Support for CAN FD frames.
- Support for Bitrate Switching (BRS) for data phase.
- Support for extended (29-bit) frame identifiers.
- Support for a bus with mixed Classic/FD nodes.
- Return the matching filter ID and type of received messages.
- Fix the maximum range for the nominal and data bit timing parameters.
- Optimize the Message RAM usage per FDCAN instance.
- Document the usage and different sections of the Message RAM.
- Update CAN docs.

PR #1603 (2022-04-02): ports/stm32: Improved CAN FD support.
* Add support for CAN FD frames.
* Add support for Bitrate Switching (BRS) for data phase.
* Add support for extended (29-bit) frame identifiers.
* Add support for a bus with mixed Classic/FD nodes.
* Return the matching filter ID and type of received messages.
* Fix the maximum range for the nominal and data bit timing parameters.
* Optimize the Message RAM usage per FDCAN instance.
* Document the usage and different sections of the Message RAM.
* Update CAN docs.
* Fixes #1071
commit 9486d168 (2022-04-02): ports/stm32: Improved CAN FD support.

* Add support for CAN FD frames.
* Add support for Bitrate Switching (BRS) for data phase.
* Add support for extended (29-bit) frame identifiers.
* Add support for a bus with mixed Classic/FD nodes.
* Return the matching filter ID and type of received messages.
* Fix the maximum range for the nominal and data bit timing parameters.
* Optimize the Message RAM usage per FDCAN instance.
* Document the usage and different sections of the Message RAM.
* Update CAN docs.
* Fixes #1071
closed 2022-04-02 by iabdalkader