openmv/openmv #1121: RPC module not working for python3

gabrielcolon32, 2021-01-15:
Hi! I've recently discussed in the forums how I tried the RPC library on Raspbian on my Raspberry Pi3B+ and it did not work. The issue was that the pixels were not being grabbed and sent from my camera to my Raspberry Pi. I tried running the examples on both python2 and python3, but it only worked on python2. I checked the camera and it was working fine on its own, but it seems that there is an issue with the RPC module.

The scripts I used for the testing were the examples in the IDE.

I don't have that much experience so I cannot give a lot of details.

I'll link the forum page: https://forums.openmv.io/viewtopic.php?p=13544#p13544

kwagyeman, 2021-01-29:
I'm finally going to get to this. Sorry for the delay. I have time now.

kwagyeman, 2021-01-29:
Hi, I just ran the scripts and they basically work. There is an issue with pygame on the PI that needs to be fixed. I will send a PR for that.

Anyway, you should see the below when you run the script:

[image]

After that I type /dev/ttyACM0.

Then when I press enter I get this:

[image]

And I also repeated the process using python3 and it worked too.

kwagyeman, 2021-01-29:
Note that I cloned the openmv repo onto my Pi and then just ran the code in the RPC folder.

kwagyeman, 2021-01-29:
https://github.com/openmv/openmv/pull/1147

closed 2021-01-29 by kwagyeman