openmv/openmv #1152: TV shield have problem with GRAYSCALE mode in 3.9.2

kidswong999, 2021-02-02:
**Describe the bug**
TV shield screen will be incorrect with grayscale in 3.9.2.

code:
```
import sensor, image, tv

sensor.reset() # Initialize the camera sensor.
sensor.set_pixformat(sensor.RGB565) # or sensor.GRAYSCALE
sensor.set_framesize(sensor.QVGA)
tv.init() # Initialize the tv.
tv.channel(8) # For wireless video transmitter shield
while(True):
    tv.display(sensor.snapshot()) # Take a picture and display the image.
```

**Please complete the following information**
 - IDE version: 2.6.8
 - Firmware version: 3.9.2
 - Board: OPENMV-H7
 - OS: Windows

kwagyeman, 2021-02-02:
@kidswong999 That's not going to display grayscale. That's color still. I put in a code path for it to display grayscale or RGB565. You need to pass a GRAYSCALE image for grayscale.

kwagyeman, 2021-02-02:
I put in code to handle grayscale:

https://github.com/openmv/openmv/blob/master/src/omv/ports/stm32/modules/py_tv.c#L557

closed 2021-02-02 by kwagyeman

kwagyeman, 2021-02-02:
Note, if you are concerned about scaling that's supported now. Just do: x_scale=tv.width(), y_scale=tv.height() in the display call. This will automatically resize the image to fix the display. Pass hint=image.BILINEAR for bilinear scaling.