openmv/openmv #1155: Script aborts when writing to 'D' pins on Portenta

idohalamit, 2021-02-08:
**Describe the bug**
The script aborts and the board requires a reset.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a pyb.Pin object with id either "D14" or "D13" (maybe other "D" pins cause the bug too, didn't check).

**Expected behavior**
I'd assume a compilation error should occur, but maybe it should interpret it the same as "PA9" and PA10" respectively.
(I'm referencing [this pinout](https://content.arduino.cc/assets/Pinout-PortentaH7_latest.png) of the portenta)

**Screenshots**
[image]

 - IDE version: 2.6.9
 - Firmware version: 3.9.2
 - Board: Portenta
 - OS: Windows

kwagyeman, 2021-02-08:
Mmm, you should have just gotten and error and nothing else should happen. The valid pins are here:

https://github.com/openmv/micropython/blob/686b4fa6d2e95e6da772062eb1f1fd8b1290fa46/ports/stm32/boards/PORTENTA/pins.csv

So, it should be PD#.

idohalamit, 2021-02-08:
👍

closed 2021-02-08 by idohalamit