openmv/openmv #1156: Snapshot lightness change if trigger frequence change on MT9V034

rongith, 2021-02-08:
- Firmware version: 3.9.2
- Board: OPENMV-H7+ MT9V034

I don't understand why if triggered mode on MT9V034 is activated, the lightness of the image change despite a fixed exposure set.
`import sensor, image, time
from pyb import Pin, ExtInt

roi = (196 , 82 , 229 , 152) #VGA f2.8
sensor.reset()
sensor.set_pixformat(sensor.GRAYSCALE)
sensor.set_framesize(sensor.VGA)
sensor.set_auto_gain(False)
sensor.set_auto_exposure(False, exposure_us=100)
sensor.set_windowing(roi)
if (sensor.get_id() ==  sensor.MT9V034):
    sensor.ioctl(sensor.IOCTL_SET_TRIGGERED_MODE, True)
sensor.skip_frames(time = 1000)
clock = time.clock()

i = 0
lasttriggertime = time.ticks_us()
cap = 0

def callback (line):
    global cap
    cap = 1

ext = ExtInt(Pin('P9'), ExtInt.IRQ_RISING, Pin.PULL_DOWN, callback)

while(True):
    if (cap == 1):
        clock.tick()
        triggertime = time.ticks_us()
        img = sensor.snapshot()
        aftersnapshottime = time.ticks_us()
        print( "snapshot " + str(i) + \
            " Cycle(ms) " + str(round(time.ticks_diff(triggertime,lasttriggertime)/1000,1)) + \
            " Snapshot(ms) " + str(round(time.ticks_diff(aftersnapshottime,triggertime)/1000,1)) + \
            " FPS " + str(round(clock.fps(),1)) )
        i = i + 1
        lasttriggertime = triggertime
        cap = 0`

Without `sensor.ioctl(sensor.IOCTL_SET_TRIGGERED_MODE, True)`
[image]
With `sensor.ioctl(sensor.IOCTL_SET_TRIGGERED_MODE, True)` and 2hz
[image]
With `sensor.ioctl(sensor.IOCTL_SET_TRIGGERED_MODE, True)` and 5hz
[image]
With `sensor.ioctl(sensor.IOCTL_SET_TRIGGERED_MODE, True)` and 10hz
[image]
With `sensor.ioctl(sensor.IOCTL_SET_TRIGGERED_MODE, True)` and 50hz
[image]

Code to set the trigger from an Arduino Uno on PIN13:
`static const uint8_t pinTrigger = 13; // The pin location of the trigger. State repeated to the PCB LED "L"
float CPS = 50; // Hz
float TotalDuration = 1000000.0 / CPS;
unsigned long ulHIGHDuration = 5000;
unsigned long ulLastTriggerEnd;

void setup() {
  pinMode(pinTrigger, OUTPUT);        // initialize pinTrigger
  Serial.begin(115200);               // Open serial communications and wait for port to open
  ulLastTriggerEnd = micros();
  Serial.println("******************************");
  Serial.print("PERIOD(ms): "); Serial.println(TotalDuration/1000.0,1);
  Serial.print("LOW(ms): "); Serial.println((TotalDuration - ulHIGHDuration)/1000.0,1);
  Serial.print("HIGH(ms): "); Serial.println(ulHIGHDuration/1000.0,1);
  //Serial.print("LastTrigger: "); Serial.println(ulLastTriggerEnd);
}

void loop() {
  //Serial.println(micros()- ulLastTriggerEnd);
  
  if ((micros()- ulLastTriggerEnd) > (TotalDuration - ulHIGHDuration)) {
    //delay(ulTotalDuration - ulHIGHDuration);
    digitalWrite(pinTrigger, HIGH);
    delay(ulHIGHDuration/1000.0);
    digitalWrite(pinTrigger, LOW);
    ulLastTriggerEnd = micros();
  }

  // If serial data received
  if (Serial.available() > 0) {
    // read the incoming byte:
    TotalDuration = Serial.parseInt();
    if (TotalDuration) {
      Serial.print("Change duration to: "); Serial.println(TotalDuration);
    }
  }  
}`

kwagyeman, 2021-03-14:
Hi, I think found the issue. It's the black level calibration logic. It's not fully turned off and it's still running when you are taking pictures. Because of this it adjusts the black level still causing the image lightness to change. I think we need to add a global control for this because it affects other cameras too.

Do this:

sensor.__write_reg(0x47, 1)

...

That said, I still see the lightness changing in the picture slightly and a lot of shot noise appearing in the image. This may just be an issue with how the sensor works. However, it's not a firmware issue on our side. If you want this fixed you will need to get into the datasheet for the device and mess with the register settings,

closed 2021-03-14 by kwagyeman

kwagyeman, 2021-03-14:
Wait, another issue. At 27 MHz PCLK output the H7 versus the H7 plus might have different clock outputs.

In particular, the H7P likely used a 30 MHz PCLK which is out of spec of the MT9V034. Versus the H7 uses a 28 MHz clock. They both are trying to use a 27 MHz clock but our clock gen logic always drives faster than required versus lower.

I'm adjusting the clock in the code to target 26,666,666 MHz which the H7P will hit perfectly and the H7 normal will hit 28.5 MHz.

kwagyeman, 2021-03-14:
This issue happens because the PCLK is always a divisor of the system freq. On the H7/H7P that's 480/2 MHz. However, some H7 units can't run faster than 400/2 MHz which results in this issue.

kwagyeman, 2021-03-14:
Maybe the 30 MHz PCLK was really out of spec.

Also, the new code properly calculates the exposure/gain values. There were some errors before because I was not using the Aptina datasheet. The Onsemi version has much less info.