openmv/openmv #1157: OV7725 configuration for minimum power and maximum speed - clarification please?

acutetech, 2021-02-10:
This is a request for clarification on how to optimise power and speed when placing the OV7725 in power saving mode. 

My requirement is to sleep until an event, then wake, take a snapshot, and go back to sleep, while minimising both power consumption and delays. I find I can do this effectively, but what I am doing violates the documentation.

[Documentation for sensor.shutdown()](https://docs.openmv.io/library/omv.sensor.html#sensor.sensor.shutdown) says: "Puts the camera into a lower power mode than sleep (but the camera must be reset on being woken up)". My experience is that I do not need to execute sensor.reset() after sensor.shutdown(), and this is much faster. 

I find it is sufficient to do this:

1 Initialise sensor once only, like this:
```
def sensorInitialise():
    sensor.reset() 
    sensor.set_pixformat(sensor.GRAYSCALE) 
    sensor.set_framesize(sensor.VGA)
```
2 Place sensor in a low power mode, like this:
```
def sensorSleep():
    sensor.sleep(True)   
    sensor.shutdown(True) 
```
3 Enter STOP mode with machine.sleep() and wait for an event (e.g. an external hardware interrupt on a pin)

4 Awaken the sensor, like this:
```
def sensorWakeup():
    sensor.shutdown(False)
    sensor.sleep(False)
```
5 Take a picture, with sensor.snapshot() for instance.

6 Repeat, by looping to step 2 (above).

This sequence means I can leave the STOP state, take a snapshot, and return to STOP very quickly, which is excellent (In the STOP state my system draws < 6mA). Previously I was executing sensor.reset() then re-initialising the sensor after the hardware interrupt, which meant about 1s elapsed before I could take the picture (and the object had moved). 

Please confirm that my sequence above is valid. (Note that I have not conducted extensive tests yet...). If so I would suggest you update the documentation and perhaps provide a new example script (I am happy to prepare that if you like).

(I can see some of the OV7725 sensor code [here](https://github.com/openmv/openmv/blob/master/src/omv/sensors/ov7725.c) but I can't see the sensor.shutdown() code - where is it?)

Configuration:
    IDE 2.6.9
    Firmware: 3.9.2
    Hardware: H7

iabdalkader, 2021-02-15:
> My experience is that I do not need to execute sensor.reset() after sensor.shutdown(), and this is much faster.

It depends on the sensor, the OV5640 datasheet mentions clearly that registers are maintained in soft and hardware standby modes, the OV7725 datasheet doesn't really specify anything, but sounds like it's the same, so it's probably okay what you're doing.

>  I can't see the sensor.shutdown() code - where is it?
sensor.shutdown just pulls the power pin up/down, it's in sensor.c.

iabdalkader, 2021-02-15:
Right and feel free to send any examples in PRs, just add some comments at the top of the script.