openmv/openmv #1175: TV Shield Screen is zoomed in when use SIF

hesokatamim, 2021-02-17:
Hello 
when I connect the tv shield into 17inch screen and I'm using "sensor.set_framesize(sensor.SIF)" the image is zoomed in on the screen, it only works fine when I use QSIF. 
I tried "tv.display(sensor.snapshot(),x_scale=tv.width(), y_scale=tv.height(),hint=image.BILINEAR) " but the IDE show error message ('module' object has no attribute 'width') which means that **tv**.width is not recognized. 
can you please provide a a code to solve this. 

thank you very much and I wish you all the best.

 - IDE version: [2.6.9]
 - Board: [ OPENMV-H7 plus]
 - OS: [Windows 10]

kwagyeman, 2021-02-17:
Update your firmware.

kwagyeman, 2021-02-17:
Like, if width is not recognized you have old firmware.

hesokatamim, 2021-06-18:
Dear kwagyeman

when i update the firmware, the Tv doesn't show any image from the tv shield. even when i use sensor.set_framesize(sensor.QSIF) still not working, i need to downgrade the firmware to have a image on TV using QSIF only, if i used larger resolution the TV will be zoomed in too much.

kwagyeman, 2021-06-18:
When we updated the driver it just does SIF resolution.

Maldonado, look at the arguments for the display method. It supports bilinear scaling now. Just pass a scale argument and you can fit the image to the display.

closed 2021-06-18 by kwagyeman