openmv/openmv #1226: "Out of fast Frame Buffer Stack Memory!" when use img.find_rects()

vamoosebbf, 2021-03-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
```python
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information**
 - IDE version: [e.g. 2.6.8]
 - Firmware version: [e.g. 3.8.0]
 - Board: [e.g. OPENMV-H7+/Portenta]
 - OS: [e.g. Windows/Linux/iOs]

kwagyeman, 2021-03-15:
Hi, if you have the H7 Plus... you may have a model that has the H7 firmware on it instead of the plus firmware. The factory messed this up on the very first batch.

Do, this:

1. Download the firmware from here: https://github.com/openmv/openmv/releases/download/v3.9.3/firmware_v3.9.3.zip
2. Use Tools->Run Bootloader to load the OPENMV4P/fimware.bin file.
3. Things should wok afterwards.

vamoosebbf, 2021-03-16:
[image]

kwagyeman, 2021-03-16:
The H7 doesn’t have enough ram to run this at QVGA. You can only use QqVga

closed 2021-03-16 by kwagyeman

AI-Amponsah, 2024-05-29:
Hi, I am using the arduino nicla vision and i am facing this same error
Exception: Failed to load ""drowsiness_quant.tflite": Out of fast frame buffer stack memory
i used QVA and had same problem

import sensor, image, time, tf, uos, gc
#define MICROPY_READER_VFS 

# Initialize the sensor
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QQVGA)
sensor.set_windowing((240, 240))
sensor.skip_frames(time=2000)

# Load TensorFlow Lite model
net = None
try:
    net = tf.load("drowsiness_quant.tflite", load_to_fb=uos.stat('drowsiness_quant.tflite')[6] > (gc.mem_free() - (64*1024)))
except Exception as e:
    raise Exception(f'Failed to load ""drowsiness_quant.tflite": {str(e)}')

# Define the labels
labels = ['Closed', 'no_yawn', 'Open', 'yawn']

# Load Haar cascades for face and eyes
face_cascade = image.HaarCascade("frontalface", stages=25)
left_eye_cascade = image.HaarCascade("lefteye", stages=25)
right_eye_cascade = image.HaarCascade("righteye", stages=25)
mouth_cascade = image.HaarCascade("mouth", stages=25)

# Thresholds and counters
drowsiness_threshold = 5
count = 0
alarm_on = False

clock = time.clock()
while(True):
    clock.tick()
    img = sensor.snapshot()

    # Detect faces in the image
    faces = img.find_features(face_cascade, threshold=0.5, scale_factor=1.5)

    for face in faces:
        img.draw_rectangle(face, color=(255, 0, 0))
        left_eye_status = "Open"
        right_eye_status = "Open"
        mouth_status = "no_yawn"

        left_eyes = img.find_features(left_eye_cascade, threshold=0.5, scale_factor=1.5, roi=face)
        right_eyes = img.find_features(right_eye_cascade, threshold=0.5, scale_factor=1.5, roi=face)
        mouths = img.find_features(mouth_cascade, threshold=0.5, scale_factor=1.5, roi=face)

        for eye in left_eyes:
            img.draw_rectangle(eye, color=(0, 255, 0))
            roi = img.copy(roi=eye).resize(145, 145)
            input_tensor = tf.image_to_tensor(roi, tf.RGB565)
            result = net.classify(input_tensor)
            left_eye_status = labels[result[0].output().index(max(result[0].output()))]
            break

        for eye in right_eyes:
            img.draw_rectangle(eye, color=(0, 255, 0))
            roi = img.copy(roi=eye).resize(145, 145)
            input_tensor = tf.image_to_tensor(roi, tf.RGB565)
            result = net.classify(input_tensor)
            right_eye_status = labels[result[0].output().index(max(result[0].output()))]
            break

        for mouth in mouths:
            img.draw_rectangle(mouth, color=(0, 255, 0))
            roi = img.copy(roi=mouth).resize(145, 145)
            input_tensor = tf.image_to_tensor(roi, tf.RGB565)
            result = net.classify(input_tensor)
            mouth_status = labels[result[0].output().index(max(result[0].output()))]
            break

        # If both eyes are closed or mouth is yawning, start counting
        if (left_eye_status == "Closed" and right_eye_status == "Closed") or mouth_status == "yawn":
            count += 1
            img.draw_string(10, 30, f"Eyes Closed, Frame count: {count}", color=(0, 0, 255))
            # If eyes are closed or mouth is yawning for 5 consecutive frames, start the alarm
            if count >= drowsiness_threshold:
                img.draw_string(100, img.height() - 20, "Drowsiness Alert!!!", color=(0, 0, 255))
                if not alarm_on:
                    alarm_on = True
                    # Implement alarm here, e.g., activate buzzer
        else:
            img.draw_string(10, 30, "Eyes Open", color=(0, 255, 0))
            count = 0
            alarm_on = False

    print(clock.fps(), "fps")

kwagyeman, 2024-05-29:
Please ask for help support on the forums.

As for the error. You need to reduce the resolution. The error is telling you that the system can't handle that large of a resolution for that algorithm.