openmv/openmv #1246: Sensor orientation rotation by 90° for B320X320-framesize

manuelzomer, 2021-03-29:
**Describe the bug**
When using the Portenta+Vision Shield in the B320X320 resolution and using sensor.set_transpose(True) to rotate the sensor orientation by 90° some of the image's pixel are "cut" from the left side and "attached" to the right side of the image (see screenshot).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Portenta + Vision Shield
2. Set sensor.set_framesize(sensor.B320X320)
3. Set sensor.set_transpose(True)
4. See how some pixel in the video feed are attached to the right side of the image instead of the left side

**Expected behavior**
The video feed should be rotated 90° without cutting the image in half.

**Screenshots**
[image]

**Please complete the following information**
 - Firmware version: https://github.com/openmv/openmv/commit/c729ed9f334cf4378f2daa2dac6bc4e19522f337
 - Board: Vision Shield + Portenta
 - OS: Windows

kwagyeman, 2021-03-29:
Okay, will fix.

iabdalkader, 2021-03-29:
The transposed path is much slower when copying pixels and it drops lines.. This is Not likely to be fixed soon, so for now you can just add this just before `set_transposed(True)`:
```
sensor.set_framerate(15)
```

kwagyeman, 2021-03-29:
I'll queue up doing the MDMA upgrade for the camera driver sooner. This should fix the issue. MDMA can do both un-transposed and transposed pixel copying.

iabdalkader, 2021-03-30:
@kwagyeman In the transposed images path the FB is accessed col-wise which causes cache thrashing this is the main reason why lines are dropped. If DCache is disabled temporarily in the line callback for transposed images it works fines. I think this should be generalized for all boards/sensors.

Side note: Himax mirror/flip (IMAGE_ORIENTATION) register is marked CMU and needs a flush after writing, otherwise it gets stuck.

kwagyeman, 2021-03-30:
I’ll just get MDMA working for this. I already read all the docs. I know how to do it. I’ll sit down and write the code and make the PR. I have to do this anyway before double buffering or we’d blow all the cpu time moving lines.

kwagyeman, 2021-04-27:
This will be fixed with the MDMA PR coming soon.

closed 2021-04-27 by kwagyeman

sebromero, 2021-04-27:
@kwagyeman Sorcery ✨