openmv/openmv #1325: Servo class not works in firmware 4.0.0

kidswong999, 2021-05-21:
**Describe the bug**
When updated to 4.0.0, the Servo class cannot drive the servo, and the oscilloscope has no waveform.
It should be a fault caused by micropython. The Timer class is works.

bad code:
```
import time
from pyb import Servo

s1 = Servo(1) # P7
s2 = Servo(2) # P8
s3 = Servo(3) # P9

while(True):
    for i in range(1000):
        s1.pulse_width(1000 + i)
        s2.pulse_width(1999 - i)
        s3.pulse_width(1000 + i)
        time.sleep_ms(10)
    for i in range(1000):
        s1.pulse_width(1999 - i)
        s2.pulse_width(1000 + i)
        s3.pulse_width(1999 - i)
        time.sleep_ms(10)
```

good code:
```
from pyb import Pin, Timer

p = Pin('P7')
tim = Timer(4, freq=50)
ch = tim.channel(1, Timer.PWM, pin=p)
ch.pulse_width_percent(50)

while True:
    pass

```

 - IDE version: [2.6.9]
 - Firmware version: [4.0.0]
 - Board: [OPENMV-H7]
 - OS: [Windows]

iabdalkader, 2021-05-21:
Could this be related ? https://github.com/openmv/openmv/commit/daa99135bc6da54d17f50f37ae986825ba30f21b @kwagyeman

kidswong999, 2021-06-03:
This bug has not been completely resolved. When the IDE is disconnected and connected again, it will no longer work.

kidswong999, 2021-06-04:
I used the OpenMV4 H7 Plus board. firmware 4.0.1.

The servo is working for the first time after power on. But running it again will not work.

code:
```
import time
from pyb import Servo

s1 = Servo(1) # P7
s2 = Servo(2) # P8

s1.pulse_width(1000)
s2.pulse_width(1000)

while(True):
    pass
```

kidswong999, 2021-06-04:
It should be a failure to reinitialize the servo. 
The following code will not generate pwm.
```
import time
from pyb import Servo

s1 = Servo(1) # P7
s1.pulse_width(1000)

s1 = Servo(1) # P7
s1.pulse_width(2000)

while(True):
    pass
```

iabdalkader, 2021-06-04:
Okay will test it.

closed 2021-06-06 by iabdalkader

kidswong999, 2023-07-22:
Hi, I tested in version 4.4.2 and Servo still has issues. Can you check it?

commit c2918717 (2021-05-23): Fix servo timer.

* Fixes #1325
commit a9a9d32d (2021-05-23): Fix servo timer.

* Fixes #1325
PR #1338 (2021-05-23): Fix servo timer.
* Fixes #1325
PR #1349 (2021-06-06): Fix servo timer channel state.
* Fixes #1325