openmv/openmv #1403: Ov7670 compile error bug

elmagnificogi, 2021-07-21:
**Describe the bug**
i use ov7670 in openmv 4. when i compile ,it always shows the ov7670 undefine.
Also, i disable the ov7690 and add the define like this:
#define OMV_OV7670_VERSION 70
#define OMV_OV7670_CLKRC 0x00
#define OMV_ENABLE_OV7670   (1)

all in the latest master branch

finaly i found the miss compile config in omv_portconfig.mk

```
UVC_OBJ += $(addprefix $(BUILD)/$(OMV_DIR)/sensors/, \
	ov2640.o                                \
	ov5640.o                                \
	ov7690.o                                \
	ov7725.o                                \
	ov9650.o                                \
	mt9v034.o                               \
	mt9m114.o                               \
	lepton.o                                \
	hm01b0.o                                \
	gc2145.o                                \
	paj6100.o                               \
	)
```
miss the ov7670.o

i add it , it's compile ok and work now

elmagnificogi, 2021-07-21:
i have pull the request about it

iabdalkader, 2021-07-25:
merged

closed 2021-07-25 by iabdalkader