openmv/openmv #1407: The morph function causes many rectangles in the image

kidswong999, 2021-07-25:
The morph function causes many rectangles in the image. It may be an algorithm error, or it may be a transmission error.
Seems to be related to resolution

[image]

code:
```
import sensor, image, time
kernel_size = 1
kernel = [-1, -1, -1,\
          -1, +8, -1,\
          -1, -1, -1]
sensor.reset()
sensor.set_pixformat(sensor.GRAYSCALE)
sensor.set_framesize(sensor.QQQVGA)
sensor.skip_frames(10)
clock = time.clock()
while(True):
    clock.tick()
    img = sensor.snapshot()
    img.morph(kernel_size, kernel)
    print(clock.fps())
```

 - IDE version: 2.6.9
 - Firmware version: 4.0.2
 - Board: OPENMV-H7 Plus, OV5640 sensor
 - OS: Windows

kwagyeman, 2021-07-26:
Will check on it. Thanks for finding the issue.

kwagyeman, 2022-01-08:
Hi, this is related to the full MDMA offload. In order to get that working I have MDMA reading from the same line buffer written by DMA2 over and over again. Assuming MDMA is able to flush it's internal fifo to SDRAM fast enough you don't see these graphical issues. However, if there's a backup then MDMA doesn't read the line in time before the next line starts over writing it.

In less than full MDMA offload there's two line buffers which prevents the issue from happening. But, the processor now has to be interrupted per line. MDMA still offloads the processor form having to do the copy operation however.

I made the full offload easy to turn off via:

OMV_ENABLE_SENSOR_MDMA_TOTAL_OFFLOAD

So, I'll set that to zero. Unfortunately, the only fix without doing this is to slow down the camera pixclk, use a larger resolution, or increase the horizontal blanking. The combo of what works changes all over the place.

kwagyeman, 2022-01-08:
Why a single buffer for DMA2? Because MDMA can't be used in a more complex mode without a linked list struct per line to tell it what to do :/.

Anyway, the only fix for this is to tune each camera settings to not have issues.

kwagyeman, 2022-01-09:
Fixed: https://github.com/openmv/openmv/pull/1498

commit 83b198f8 (2022-01-09): * Make sure all cached CPU writes are flushed before returning the buffer.

* Should fix #1407
commit 9fb1d54d (2022-01-09): * Make sure all cached CPU writes are flushed before returning the buffer.

* Should fix #1407
commit 1c6b452b (2022-01-09): * Invalidate framebuffer cache before using.

* Fixes #1407
PR #1498 (2022-01-09): Make sure cached CPU writes are flushed before returning the buffer
@kwagyeman 

* Should fix #1407
closed 2022-01-09 by kwagyeman