openmv/openmv #1427: Portenta H7 - ‘module’ object has no attribute ‘WINC’ while using rpc library

sw4p, 2021-08-07:
I am trying to send an image from a Portenta H7 board to a PC but getting *‘module’ object has no attribute ‘WINC’* error. It is related to rpc.py library. The full error is mentioned below.

File “”, line 7, in
File “rpc.py”, line 560, in
File “rpc.py”, line 590, in rpc_wifi_master
AttributeError: ‘module’ object has no attribute ‘WINC’
MicroPython: v1.15-r51 OpenMV: v4.1.0 HAL: v1.9.0 BOARD: PORTENTA-STM32H747

I am running the *image_transfer_jpg_as_the_remote_device_for_your_computer.py* example.

similar issue - https://github.com/openmv/openmv-arduino-rpc/issues/2

**response by iabdalkader** - The WiFi module on Portenta is called WLAN, also the API is slightly different the WINC's API, see the WiFi examples for Portenta. I can make the WINC module compatible with the Portenta WiFi module (which is also the module used by official mpy boards) so the same scripts work with both modules (please open an issue). That said, this module will still need to be updated.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the *image_transfer_jpg_as_the_remote_device_for_your_computer.py* example on Portenta H7.

**Expected behavior**
Including rpc library should not throw any error.

**Please complete the following information**
 - IDE version: 2.6.9
 - Firmware version: v4.1.0 
 - Board: Portenta
 - OS: Windows

sw4p, 2021-08-07:
I guess the issue is here
[image]

updating rpc library to use Portenta's WLAN module should fix the issue or the way @iabdalkader suggested (captured in the above issue)

iabdalkader, 2021-08-08:
You can try to fix this if you want, just follow the WiFi examples for Portenta and see how the WiFi module is initialized and used and make fixes in the rpc library, if you're in rush. I'll work on making the WINC module look like the CYW43 module so that the same could can work on it, but it will take at least a couple of weeks to get around to doing another release.

kwagyeman, 2021-08-08:
The easiest way is just to try to import WINC, catch the error, and then import WLAN instead. It's a very few easy fixes since sockets are the same.

sw4p, 2021-08-09:
Thank you @kwagyeman and @iabdalkader! I will try the easy fix method first; if that works - awesome! if not then I will update the rpc library with WLAN.

iabdalkader, 2021-08-10:
> The easiest way is just to try to import WINC, catch the error, and then import WLAN instead

@kwagyeman It's actually the only the way, the RPC library still needs to support the older firmware with the old WINC API. I'm making WINC API compatible with the upstream WLAN used by the official MicroPython boards, Portenta and the Nano RP2040 so this doesn't happen again in the future, and it will also allow all boards to use the same networking examples (less docs too).

sw4p, 2021-08-12:
I have managed to make it work, however, I have only tested it in wifi_slave mode.
[image]

afridali123, 2021-09-19:
Can someone please help me how to fix this issue with steps please ?