openmv/openmv #1443: IDE Exception when extending bytearray() with byte() or str or other buffer protocols

gramss, 2021-08-30:
**Describe the bug**
I cannot extend a bytearray() with a byte, String, another bytearray() or probably the rest which supports the buffer protocol.
Also related to: https://github.com/micropython/micropython/issues/7408

This is what I get:

> Exception: IDE interrupt
> MicroPython: v1.15-r57 OpenMV: v4.1.1 HAL: v1.9.0 BOARD: OPENMV4-STM32H743

**To Reproduce**
```
import image, network, math, rpc, sensor, struct, tf

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)

img = sensor.snapshot()
triggered = 25
#t= t.to_bytes(1, "big")
#print(t)
compr = img.compress(quality=90).bytearray()
msg = compr.extend(triggered.to_bytes(2, "big")) #here is where the exception comes and the OpenMV Cam restarts/resets
print(msg)
#print(img.compress(quality=90).bytearray().extend([triggered.to_bytes(1, "big")]))
```

**Expected behavior**
I want to add an integer to an bytearray..

**Please complete the following information**
 - IDE version: 2.81
 - Firmware version: 4.1.1
 - Board: OPENMV-H7
 - OS: MacOS

gramss, 2021-08-30:
Something like this works, but is probably not as performant..?

```
import image, network, math, rpc, sensor, struct, tf

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)

img = sensor.snapshot()
triggered = 25
s = bytearray(str(triggered))
s = bytearray(triggered.to_bytes(1, "big"))
print(s)
compr = img.compress(quality=90).bytearray()
#msg = compr.extend(s)
s[0:0] = compr
print(s)
```

Regarding the usecase:
I want to add an integer or a bool to an image stream over the rpc-lib to have the possibility to stream multiple item at once.

kwagyeman, 2021-08-30:
Hi, the bytearray() method just gives you a raw pointer to the image. You cannot extend it as it's not on the heap... but, in our large memory space. MicroPython when you try to extend it would need to allocate a complete copy of the image on the heap which will generate an out of memory exception as it tries to deep copy the image. We have much less space on the heap than in the frame buffer.

You'll need to add a second RPC call to move the extra value you want to send, or, use the .compressed() method which will compress the jpeg image and then allocate it as a bytes() object on the heap that you can extend as you like. However, there might not be space.

closed 2021-12-30 by kwagyeman