openmv/openmv #1444: Uncought exception in firmware?

gramss, 2021-08-31:
Hey,

I'm playing around a bit. And I somehow managed to get the REPL duplicated on the UART line.

I often have the OpenMV H7 cam work "normally" - as in no rapid flashing and when I connect it to the IDE it just goes bonkers. (Rapid flashing of leds and IDE cannot response to it)

Gladly I catched the REPL output due to my UART line. Here it is:

```
__class__       __name__Exception: IDE interrupt
FATAL: uncaught exception 30001a64
```
As I have not a clue about the firmware currently, this error code seems to me that it could be an "easy" fix.. :)

The inital bad behavior is coming from my rpc_usb_slave activities I'm currently trying out.

The error that completly crashes (maybe internal flash and what not) was this:

```
hello Terminal
done registering callbacks
Sanity!
Traceback (most recent call last):
  File "/main.py", line 154, in <module>
  File "rpc.py", line 280, in loop
  File "/main.py", line 108, in movement_im_stream
AttributeError: 'ROS2_Interface' object has no attribute 'jpeg_image_stream_cb'
MicroPython: v1.15-r57 OpenMV: v4.1.1 HAL: v1.9.0 BOARD: OPENMV4-STM32H743
Type "help()" for more information.
>>> !a!a!a!a!a!a!a!a!a!a!a!a!a!a!a!a!a!aD
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 
__class__       __name__        LED             UART
image           math            network         sensor
struct          tf              time            usys
uart            rpc             os              ROS2_Interface
blue_led        A
>>> D
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'D' isn't defined
>>> 
__class__       __name__        LED             UART
image           math            network         sensor
struct          tf              time            usys
uart            rpc             os              ROS2_Interface
blue_led        A
>>> D
```

The last part just goes on and on. Even manages to somehow crash my UART device sometimes.
The error causing this was totally true. I had a function named falsely. "Hard" to spot without any proper REPL if I need to disconnect it due to the usage of rpc_usb.. :)

-----
Offtopic:

How I got the REPL to work as duplicate over UART:

```
import time, usys

import os
from pyb import UART

# UART 3, and baudrate.
uart = UART(3, 19200)

os.dupterm(uart)

while(True):
    #uart.write("Hello World!\n")
    #if (uart.any()):
        #print(uart.read())
    time.sleep_ms(1000)
    try:
        usys.exit(1)
    except SystemExit:  
        usys.print_exception(SystemExit)
```

metadata: latest firmware 4.1.1 and latest IDE like the issue before

kwagyeman, 2021-09-01:
Not really sure what you want me to say. You aren't supposed to use the VCP interface with the IDE.

gramss, 2021-09-01:
Okay. That explains the error. But hear me out:

I'm in the middle of programming. So I need to iterate my code until it works. I totally understand and accept that I cannot use the IDE while using rpc in USB mode. That's why I use the "save currently open file as main.py" and reset the cam via IDE (-> device get's disconnected).
Now I tested my interface, experienced an error, optionally saw what is wrong through the UART, stop my testing program using the VCP and **NOW**: I want to reconnect my cam back to the IDE and try to find & fix the error in code.

The issue here is, that if the cam hard fails, I cannot connect to it. I first would need to erase the internal flash or the SD card to stop the programm from running. Or be fast enough in the IDE to connect during bootup..?

Possible fix:
Catch exception and do a `usys.SystemExit()` + optionally some user notification like the one I get when I start the code in the IDE and it tells me to close everything. This is probably something I just to do myself by wrapping my whole code into a try/exception . Probably nothing that should be included in of the MV-cam firmware?

kwagyeman, 2022-01-09:
Hi, you have to click the erase flash button in the IDE and then remove and reconnect your OpenMV Cam. This will flush the internal flash.

Alternatively, you can use an SD card with the script on it.

>> Possible fix:
Catch exception and do a usys.SystemExit() + optionally some user notification like the one I get when I start the code in the IDE and it tells me to close everything. This is probably something I just to do myself by wrapping my whole code into a try/exception . Probably nothing that should be included in of the MV-cam firmware?

Can you make a feature request for an at exit method to MicroPython?

kwagyeman, 2022-01-09:
I know the whole control via the PC and VCP is bad right now. MicroPython does support multiple VPC ports. However, it's a lot of work to get this working in our firmware nicely. @iabdalkader - It would be nice to have a dedicated debug UART for connection to the IDE and another for the PC...

closed 2022-01-09 by kwagyeman