openmv/openmv #1452: module’ object has no attribute ‘WINC while running rtsp code

afridali123, 2021-09-19:
module’ object has no attribute ‘WINC,

Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
  File "rtsp.py", line 10, in <module>
  File "rtsp.py", line 40, in rtsp_server
AttributeError: 'module' object has no attribute 'WINC'
MicroPython: v1.15-r57 OpenMV: v4.1.1 HAL: v1.9.0 BOARD: PORTENTA-STM32H747
Type "help()" for more information.

I'm running the "rtsp video stream example", i got this below error.

i've already seen the issue of https://github.com/openmv/openmv/issues/1427

i didnt get how to solve it, please help me in solving issue with steps, thanks

[image]

afridali123, 2021-09-19:
@iabdalkader

kwagyeman, 2021-09-19:
It's because the libraries were written for our WINC module and not the Portenta. So, the wifi adapter code is wrong. You can fix by changing that said adapter code. I'll try to get on this soon.

afridali123, 2021-09-19:
ok thanks @kwagyeman , waiting for your reply, pl help me what portion code i need to change to run this example, thank you somuch in advance

kwagyeman, 2021-09-19:
You need to take the code here and save it to the uPy disk. https://github.com/openmv/openmv/blob/master/scripts/libraries/rtsp.py#L40

Modify the init method to use the wlan module versus winc. The socket code stays the same.

For some reason VLC doesn't like our MJPEG format. I need to fix this. However, you can use ffpmeg to display video. You can point it to the rstp url and it will play the video.

afridali123, 2021-09-19:
looks like i need to edit the rtsp lib file. what is meant by save to uPY disk ?

please help me with steps or share some links to be done, thank you in advance

afridali123, 2021-09-21:
@kwagyeman @iabdalkader can you pl help me in this regard

iabdalkader, 2021-09-21:
I updated the WINC API to be compatible with CYW/WLAN, there's nothing else I can do the RPC library need to be fixed @kwagyeman

kwagyeman, 2021-10-17:
Fixed in: https://github.com/openmv/openmv/pull/1460

closed 2021-10-17 by kwagyeman