openmv/openmv #1465: MLC and Nano RP2040 Not Working

canchebagur, 2021-11-08:
I tried to run the MLC examples on the Arduino Nano RP2040, specifically the [LSM6DSOX MLC example](https://github.com/openmv/openmv/blob/master/scripts/examples/Arduino/Nano-RP2040/01-Sensors/lsm6dsox_mlc.py), but I keep getting the following error:

Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
TypeError: unexpected keyword argument 'gyro_odr'
MicroPython: v1.15-r57-dirty OpenMV: v4.1.1-dirty HAL: v1.2.0 BOARD: Arduino Nano RP2040 Connect-RP2040
Type "help()" for more information.

Is there something I am missing? Thanks in advance for the help!

 - IDE version: 2.8.1
 - Firmware version: 4.1.1
 - Board: Arduino Nano RP2040 Connect 
 - OS: Windows

iabdalkader, 2021-11-09:
This feature is not released yet, you need to be build the firmware from source or wait for the next release which will be out soon.

canchebagur, 2021-11-09:
Thanks for the answer @iabdalkader. To build the latest firmware for the NanoRP2040 Connect what should be passed in the compiler?

**make TARGET=NANORP2040** ???

iabdalkader, 2021-11-09:
```TARGET=ARDUINO_NANO_RP2040_CONNECT```

iabdalkader, 2021-11-09:
The new release is out, you can just use the compiled firmware.

closed 2021-11-09 by iabdalkader