openmv/openmv #1468: edge impulse model crash in 4.1.2

kidswong999, 2021-12-06:
Code: https://github.com/SingTown/mask-tflite
Board: OpenMV4 H7 Plus

The code crashes in V4.1.2, but works well in V4.1.1. 

The IDE disconnects directly without any error prompts.

I re-generated the model from edge impulse, the same fault.

I think there's something wrong with tf.classify.

iabdalkader, 2021-12-06:
@kwagyeman This commit db43f98b77f7ded3b9be570008be79b05300fe56 broke something in TF, if you revert it and rebuild it works just fine. It could also be why the TF update wasn't working (that issue you were debugging).

To test without it:
```bash
git checkout -b tf_test
git revert db43f98b77f7ded3b9be570008be79b05300fe56
make -j8 TARGET=OPENMV4P
```

kwagyeman, 2021-12-06:
Okay, will fix tonight. Should be able to figure out what's wrong. Maybe this is why the new TF doesn't work.

kwagyeman, 2021-12-12:
https://github.com/openmv/openmv/pull/1472

closed 2021-12-12 by kwagyeman