openmv/openmv #1502: Edge Impulse Model Class result mistake on v4.2.1

kidswong999, 2022-01-13:
code and model:https://github.com/SingTown/mask-tflite

The result of the classification is wrong and sum is Not 1.
Maybe the tf library is causing the problem.

Result:

```
**********
Predictions at [x=0,y=0,w=240,h=240]
face = 0.500000
mask = 0.500000
8.03296 fps
**********
Predictions at [x=0,y=0,w=240,h=240]
face = 0.675781
mask = 1.324219
8.03374 fps
**********
Predictions at [x=0,y=0,w=240,h=240]
face = 0.515625
mask = 1.484375
8.03606 fps
**********
Predictions at [x=0,y=0,w=240,h=240]
face = 0.835938
mask = 1.164063
8.03674 fps

```

kwagyeman, 2022-01-13:
Hi, yeah, I changed the scaling output to match what the networks actually want and it's not 1 anymore.

I will bug edge impulse folks about this. They told me to modify the scaling and offset to fix some networks and now folks are telling me this broke things.

closed 2022-02-17 by iabdalkader

iabdalkader, 2022-02-17:
@kidswong999 This is fixed now.

PR #1540 (2022-02-17): py_tf.c: Fix output scaling for uint8/int8.
Fixes  #1502