openmv/openmv #1505: PICO/RP2 allocates more vbuffers than available.

iabdalkader, 2022-01-18:
**To Reproduce**
This is hard to reproduce without my custom sensor kit, but I could help with testing.

**Expected behavior**
Expected to allocate 1 buffer.

**Screenshots**
[image]

**Please complete the following information**
 - IDE version: [latest]
 - Firmware version: [latest]
 - Board: [PICO, NANO-RP2040]
 - OS: [Linux]

kwagyeman, 2022-01-19:
Can they send me this board?

kwagyeman, 2022-01-19:
Maybe this? https://github.com/openmv/openmv/pull/1506

kwagyeman, 2022-01-19:
I'd put printf here:

`void framebuffer_auto_adjust_buffers()
{
     // Keep same buffer count in video fifo mode but resize buffer sizes.
    if (framebuffer->n_buffers > 3) {
        framebuffer_set_buffers(framebuffer->n_buffers);
        return;
    }

    for (int i = 3; i > 0; i--) {
        framebuffer_set_buffers(i);

        // Find a buffering size automatically that doesn't use more than half.
        if (fb_avail() >= framebuffer_total_buffer_size()) {
            return;
        }
    }
}`

And just print out what the numbers are each loop. Issue should be obvious.

iabdalkader, 2022-01-19:
> Can they send me this board?

This my own custom breakout for Pico. I'll try the above later.

commit c2b6e2a1 (2022-01-19): NANO-RP2040: Fix raw buffer size.

* Fixes #1505
PR #1507 (2022-01-19): NANO-RP2040: Fix raw buffer size.
* Fixes #1505
closed 2022-01-19 by iabdalkader