openmv/openmv #1508: pydfu.py always dies with "Pipe error" when trying to program STM32F072

ned-pcs, 2022-01-21:
**Describe the bug**
`pydfu.py` always dies with "Pipe error" when trying to program STM32F072.
I can program the chip without errors using `dfu-util`.

**To Reproduce**
Connect USB to an STM32F072 in DFU mode.
Try to use `pydfu.py` to transfer a file to the device:
```
pydfu.py -u my_program.dfu -v -v
```
Get the following result:
```
File: my_program.dfu
    DfuSe v1, image size: 54461, targets: 1
    Target 0, alt setting: 0, name: "ST...", size: 54160, elements: 1
      0, address: 0x08000000, size: 54152
    usb: 0483:df11, device: 0x0000, dfu: 0x011a, UFD, 16, 0xbbe27d8e
Writing memory...
0x08000000   54152 [                         ]   0% Erasing page: 0x8000000...
Addr 0x8000000 0KBs/2KBs...
0x08000000   54152 [                         ]   0% Traceback (most recent call last):
  File "./pydfu.py", line 582, in <module>
    main()
  File "./pydfu.py", line 570, in main
    write_elements(elements, args.mass_erase, progress=cli_progress)
  File "./pydfu.py", line 475, in write_elements
    elem_addr, elem_size)
  File "./pydfu.py", line 200, in write_memory
    if get_status() != __DFU_STATE_DFU_DOWNLOAD_IDLE:
  File "./pydfu.py", line 114, in get_status
    stat = __dev.ctrl_transfer(0xA1, __DFU_GETSTATUS, 0, __DFU_INTERFACE, 6, 20000)
  File "/home/ned/.local/lib/python2.7/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/home/ned/.local/lib/python2.7/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/home/ned/.local/lib/python2.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 32] Pipe error
```

**Expected behavior**
I expected the DFU programming to complete.
`dfu-util` works OK on both x86_64 and aarch64 systems, giving the following output:
```
% dfu-util -D my_program.dfu -v
dfu-util 0.11

libusb version 1.0.24 (11584)
DFU suffix version 11a
Match vendor ID from file: 0483
Match product ID from file: df11
Multiple alternate interfaces for DfuSe file
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
DFU attributes: (0x0b) bitCanDnload bitCanUpload bitWillDetach
Detach timeout 255 ms
Claiming USB DFU Interface...
Setting Alternate Interface #1 ...
Determining device status...
DFU state(5) = dfuDNLOAD-IDLE, status(0) = No error condition is present
Aborting previous incomplete transfer
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Option Bytes  "
dfu-util: Non-valid multiplier ' ', assuming bytes
Memory segment at 0x1ffff800   1 x   16 =    16 (rw)
DfuSe interface name: "Internal Flash  "
Memory segment at 0x08000000  64 x 2048 = 131072 (rew)
File contains 1 DFU images
Parsing DFU image 1
Target name: ST...
Image for alternate setting 0, (1 elements, total size = 54160)
Setting Alternate Interface #0 ...
Parsing element 1, address = 0x08000000, size = 54152
Erasing page size 2048 at address 0x08000000, page starting at 0x08000000
Erasing page size 2048 at address 0x08000800, page starting at 0x08000800
Erasing page size 2048 at address 0x08001000, page starting at 0x08001000
Erasing page size 2048 at address 0x08001800, page starting at 0x08001800
Erasing page size 2048 at address 0x08002000, page starting at 0x08002000
Erasing page size 2048 at address 0x08002800, page starting at 0x08002800
Erasing page size 2048 at address 0x08003000, page starting at 0x08003000
Erasing page size 2048 at address 0x08003800, page starting at 0x08003800
Erasing page size 2048 at address 0x08004000, page starting at 0x08004000
Erasing page size 2048 at address 0x08004800, page starting at 0x08004800
Erasing page size 2048 at address 0x08005000, page starting at 0x08005000
Erasing page size 2048 at address 0x08005800, page starting at 0x08005800
Erasing page size 2048 at address 0x08006000, page starting at 0x08006000
Erasing page size 2048 at address 0x08006800, page starting at 0x08006800
Erasing page size 2048 at address 0x08007000, page starting at 0x08007000
Erasing page size 2048 at address 0x08007800, page starting at 0x08007800
Erasing page size 2048 at address 0x08008000, page starting at 0x08008000
Erasing page size 2048 at address 0x08008800, page starting at 0x08008800
Erasing page size 2048 at address 0x08009000, page starting at 0x08009000
Erasing page size 2048 at address 0x08009800, page starting at 0x08009800
Erasing page size 2048 at address 0x0800a000, page starting at 0x0800a000
Erasing page size 2048 at address 0x0800a800, page starting at 0x0800a800
Erasing page size 2048 at address 0x0800b000, page starting at 0x0800b000
Erasing page size 2048 at address 0x0800b800, page starting at 0x0800b800
Erasing page size 2048 at address 0x0800c000, page starting at 0x0800c000
Erasing page size 2048 at address 0x0800c800, page starting at 0x0800c800
Erasing page size 2048 at address 0x0800d000, page starting at 0x0800d000
 Download from image offset 00000000 to memory 08000000-080007ff, size 2048
 Download from image offset 00000800 to memory 08000800-08000fff, size 2048
 Download from image offset 00001000 to memory 08001000-080017ff, size 2048
 Download from image offset 00001800 to memory 08001800-08001fff, size 2048
 Download from image offset 00002000 to memory 08002000-080027ff, size 2048
 Download from image offset 00002800 to memory 08002800-08002fff, size 2048
 Download from image offset 00003000 to memory 08003000-080037ff, size 2048
 Download from image offset 00003800 to memory 08003800-08003fff, size 2048
 Download from image offset 00004000 to memory 08004000-080047ff, size 2048
 Download from image offset 00004800 to memory 08004800-08004fff, size 2048
 Download from image offset 00005000 to memory 08005000-080057ff, size 2048
 Download from image offset 00005800 to memory 08005800-08005fff, size 2048
 Download from image offset 00006000 to memory 08006000-080067ff, size 2048
 Download from image offset 00006800 to memory 08006800-08006fff, size 2048
 Download from image offset 00007000 to memory 08007000-080077ff, size 2048
 Download from image offset 00007800 to memory 08007800-08007fff, size 2048
 Download from image offset 00008000 to memory 08008000-080087ff, size 2048
 Download from image offset 00008800 to memory 08008800-08008fff, size 2048
 Download from image offset 00009000 to memory 08009000-080097ff, size 2048
 Download from image offset 00009800 to memory 08009800-08009fff, size 2048
 Download from image offset 0000a000 to memory 0800a000-0800a7ff, size 2048
 Download from image offset 0000a800 to memory 0800a800-0800afff, size 2048
 Download from image offset 0000b000 to memory 0800b000-0800b7ff, size 2048
 Download from image offset 0000b800 to memory 0800b800-0800bfff, size 2048
 Download from image offset 0000c000 to memory 0800c000-0800c7ff, size 2048
 Download from image offset 0000c800 to memory 0800c800-0800cfff, size 2048
 Download from image offset 0000d000 to memory 0800d000-0800d387, size 904
Done parsing DfuSe file
```

**Please complete the following information**
 - IDE version: none
 - Firmware version: none
 - Board: none
 - OS: Linux on x86_64 and aarch64

I have tried the `pydfu.py` on a MicroPython PyBoard v1.1 (STM32F405) and it works fine.
According to ST, both the STM32F072 and the STM32F405 have the same version (2.2) of the DFU.

Thanks for writing this utility!

iabdalkader, 2022-01-31:
> According to ST, both the STM32F072 and the STM32F405 have the same version (2.2) of the DFU.

This is not always true, they say it's compatible but then make the subtlest change that breaks it. For example they changed something in H7 rev V DFU that broke `dfu-util`. I have no way of testing or fixing this, and we no longer use this tool to be honest, so I'm not really interested in investigating it further. I'd try to use Wireshark to see what `dfu-util` is doing differently if I were you and try to fix it, you're welcome to send PRs.

kwagyeman, 2024-01-25:
Closing this old issue. pydfu is not used anymore in favor of dfu-util.

That said, as a note, the program is still distributed by OpenMV IDE and available as a fallback in case dfu-util fails. However, we only use it to load the first sector of any stm32 device per what Ibrahim mentioned above. On rev V silicon of the H7 ST broke pydfu and dfu-util for accessing anything beyond the first couple of sectors of their chips. You cannot depend on dfu programming being reliable for ST unless you use their cube-mx tool which is impossibly large.

closed 2024-01-25 by kwagyeman