openmv/openmv #1517: Portenta default i2c settings break the DCMI

lestofante, 2022-01-31:
**Describe the bug**
The alternate i2c for the Portenza board has been added as i2c4 (https://github.com/openmv/openmv/pull/1399) but this is sharing the hardware pin with the DCMI.  
Another PR then made the alternate i2c used by default (https://github.com/openmv/openmv/commit/f4a6c4003b8d6f1ba6fc3ade9f8a8c9d957053fe)  
As far as i understand, this mean now we have a paradox where DCMI cameras wont work anymore by default, but they are enabled by default.

**Expected behavior**
As quick fix, enabling a DCMI camera + i2c4 should generate compilation error.
A better fix may be an easy way to select the default i2c; my suggestion is to change the #ifdef ISC_I2C_ALT in https://github.com/openmv/openmv/commit/f4a6c4003b8d6f1ba6fc3ade9f8a8c9d957053fe with an #ifdef USE_I2C_ALT or something, that can be feed at compilation time.
Or maybe have two configuration, one for the Portenta and one for the Portenta + Arduino carrier board.

**Please complete the following information**
 - Firmware version: master 9cedae8
 - Board: OPENMV-H7+/Portenta

iabdalkader, 2022-01-31:
This is bus used on the carrier board, and yes it is in conflict with DCMI pins, but it's never scanned unless the sensor wasn't detected on the first bus. This will be fixed when the carrier board is revised, but for now it's not causing any issues.

lestofante, 2022-01-31:
Thanks, I will try and report if successful, I think I changed the code and patched the board at the same time, so ill try with the original firmware.