openmv/openmv #1536: OpenMV Pure Thermal  RTC

weilian1977, 2022-02-16:
OpenMV Pure Thermal，connected a battery to VBATT，VBATT_RTC can output 3.0V even when power off，but when power off RTC clock can only keep the time of power off time while not increment。firmware 4.2.1

iabdalkader, 2022-02-17:
This should be fixed now.. Try [v4.2.3](https://github.com/openmv/openmv/releases/tag/v4.2.3) ? When you start/stop a script it should print in the terminal `OpenMV v4.2.3`  to be sure you have the right firmware.

weilian1977, 2022-02-18:
Now I confirm  firmware is v4.2.3，test result is still keep the time while not increment。
The test method is connect a battery(3.6V) to VBATT,then connect Pure Thermal board  connect Pure Thermal board to computer through usb cable,power on Pure Thermal board,set RTC time,power off Pure Thermal board while keep usb cable connect Pure Thermal board and computer ,wait about 10min,power on Pure Thermal board,get RTC time.

weilian1977, 2022-02-18:
read  RCC_BDCR Register   hex(stm.mem32[0x58024400+0x70]) is 0x8203, 8 means  RTC clock enable，2 means RTC clock source selection is LSI，3 means that LSE oscillator ready and LSE oscillator enabled, so RTC clock source  is LSI.

iabdalkader, 2022-02-19:
@weilian1977 is the issue fixed ?

weilian1977, 2022-02-21:
> Yes that's right.. I pushed another fix, please try the latest development release again.

https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip  V4.2.4  print(hex(stm.mem32[0x58024400+0x70]))=0x8200    not increase

https://github.com/openmv/openmv/releases/download/v4.2.3/firmware_v4.2.3.zip  
V4.2.3  print(hex(stm.mem32[0x58024400+0x70]))=0x8200    not increase
disconnect battary then power on  
time  is (2015, 1, 1, 4, 0, 0, 7, 222656)
 
print(hex(stm.mem32[0x58024400+0x70]))=0x8200

iabdalkader, 2022-02-21:
This is not the development release. Please try this: https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip

weilian1977, 2022-02-22:
download（https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip），StmCubeProgrammer download the openmv.bin，IDE show version is 4.2.4，print(hex(stm.mem32[0x58024400+0x70]))=0x8205，no increase，test C96 C97 C110 C111，there are on 32.768 signal。
print(rtc.datetime())=(2022, 2, 22, 2, 8, 10, 35, 605468).

weilian1977, 2022-02-22:
Need to disconnect the battery after burning the firmware？

weilian1977, 2022-02-22:
download（[https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip），StmCubeProgrammer download the openmv.bin，disconnect the battery after burning the firmware about 5 min，IDE show version is 4.2.4，print(hex(stm.mem32[0x58024400+0x70]))=0x8200，no increase，test C96 C97 C110 C111，there are on 32.768 signal。

iabdalkader, 2022-02-22:
> disconnect the battery after burning the firmware about 5 min

You **should never disconnect the battery**, if you want the RTC to keep working it needs a power source.

weilian1977, 2022-02-22:
when I test RTC，I never disconnect the battery。

iabdalkader, 2022-02-22:
Okay I see an issue, wait will update the dev release.

iabdalkader, 2022-02-22:
Hi, try it again now, it should be working. I made sure RTC is using LSE now.

https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip

weilian1977, 2022-02-22:
the firmware can not power on OPENMVPT，I download it two times and burning the firmware  two times

iabdalkader, 2022-02-23:
Will debug and get back to you.

iabdalkader, 2022-02-26:
@weilian1977 The wrong LSE mode was used, I fixed it now and the RTC clock source should LSE. Please test again and let me know how it goes.
Same link: https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip

iabdalkader, 2022-02-28:
Hi, the firmware should be working, but there's an issue with the LSE crystal. The crystal load caps are too high so the LSE doesn't start, and the firmware blocks in a loop waiting for the LSE to start. I disabled the early LSE/RTC config, so the firmware shouldn't block anymore, and left the RTC clock source in Micropython set to LSE. However, the RTC will Not work unless capacitors `C96` and `C97` are removed.

weilian1977, 2022-03-01:
Thanks a lots.I will remove C96 and C97(noly C96   C97?),which firmware  can test?

iabdalkader, 2022-03-01:
Same link above, development release. It gets updated after every change. Please let us know how it goes.

weilian1977, 2022-03-01:
OK！

iabdalkader, 2022-03-01:
For reference, these are the capacitors that need to be removed:

[image]

weilian1977, 2022-03-02:
1、capacitors was removed；
2、firmware is  https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip
3、power off openmvPT and disconnect battery；
4、connect battery；
5、power on openmvPT by usb；
6、rtc.datetime((2022, 3, 2, 2, 14, 56, 00, 0)) set time 
7、print(hex(stm.mem32[0x58024400+0x70]))  == 0x81038、power off openmvPT by SW1（PWR ON/OFF ） in PTOMV_Schematics_revE_prelim.PDF page2  while connect battery；9、power on openmvPT by SW1（PWR ON/OFF ） in PTOMV_Schematics_revE_prelim.PDF page2  while connect battery；10、print(rtc.datetime())= 0:0:3011、print(hex(stm.mem32[0x58024400+0x70]))  == 0x8203

iabdalkader, 2022-03-03:
> 11、print(hex(stm.mem32[0x58024400+0x70]))  == 0x8203

Does the RTC increment ? I see the RTC clock source is switched back to LSI on reset, I pushed another fix. Please test again with the same link above.

weilian1977, 2022-03-03:
power off （ RTC battery is connect ）will cause RTC clock source from lse switched LSI，and time to 0：0：0：

iabdalkader, 2022-03-03:
Did you test again ? I made sure the LSE source is not changed on reset.

weilian1977, 2022-03-03:
need new firmware？

iabdalkader, 2022-03-03:
Yes, same link it gets updated. https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip

weilian1977, 2022-03-03:
1、capacitors was removed；
2、firmware is  https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip
       info:MicroPython: 188f051 OpenMV: 68eb6ca HAL: v1.9.0 BOARD: OPENMVPT-STM32H743
3、power off openmvPT and disconnect battery；
4、connect battery；
5、power on openmvPT by usb；
6、rtc.datetime((2022, 3, 3, 4, 18, 58, 00, 0)) set time   (set time code and result)

7、print(hex(stm.mem32[0x58024400+0x70]))  = 0x8103   (set test code code and result1)8、power off openmvPT by SW1（PWR ON/OFF ） in Schematics_revE_prelim.PDF page2  while connect battery；9、power on openmvPT by SW1（PWR ON/OFF ） in Schematics_revE_prelim.PDF page2  while connect battery；10、print(rtc.datetime())= 2000, 1, 1, 1, 0, 0, 31, 734375    (set test code code and result2)11、print(hex(stm.mem32[0x58024400+0x70]))  == 0x8203    (set test code code and result2)

set time codeimport timefrom pyb import RTCrtc = RTC()rtc.datetime((2022, 3, 3, 4, 18, 58, 00, 0))while (True):    print(rtc.datetime())    time.sleep_ms(1000)
result:
(2022, 3, 3, 4, 18, 58, 43, 191406)

0x8103

test codeimport timefrom pyb import RTCimport stmrtc = RTC()#rtc.init()#stm.mem32[0x58024400+0x70]=0x8103print(hex(stm.mem32[0x58024400+0x70]))print(hex(stm.mem32[0x58024800]))while (True):    print(rtc.datetime())    print(hex(stm.mem32[0x58024400+0x70]))    time.sleep_ms(1000)result2:
0x8103

-0xfff3f00

(2022, 3, 3, 4, 18, 58, 38, 335937)

0x8103

(2022, 3, 3, 4, 18, 58, 39, 304687)

0x8103

(2022, 3, 3, 4, 18, 58, 40, 277343)

result3:
0x8203

-0xfff3f00

(2000, 1, 1, 1, 0, 0, 28, 847656)

0x8203

(2000, 1, 1, 1, 0, 0, 29, 808593)

0x8203

(2000, 1, 1, 1, 0, 0, 30, 769531)

0x8203

weilian1977, 2022-03-03:
1、capacitors was removed；
2、firmware is  https://github.com/openmv/openmv/releases/download/development/firmware_OPENMVPT.zip
       info:MicroPython: 188f051 OpenMV: 68eb6ca HAL: v1.9.0 BOARD: OPENMVPT-STM32H743
3、power off openmvPT and disconnect battery；
4、connect battery；
5、power on openmvPT by usb；
6、rtc.datetime((2022, 3, 3, 4, 18, 58, 00, 0)) set time   (set time code and result)

7、print(hex(stm.mem32[0x58024400+0x70]))  = 0x8103   (test code code and result)8、power off openmvPT by SW1（PWR ON/OFF ）9、power on openmvPT by SW1（PWR ON/OFF ）10、print(rtc.datetime())= 2000, 1, 1, 1, 0, 0, 31, 734375    (test code code and result2)11、print(hex(stm.mem32[0x58024400+0x70]))  == 0x8203    (test code code and result3)
12、rtc.datetime((2022, 3, 3, 4, 19, 13, 20, 0))  set time   (set time code and result)
13、print(hex(stm.mem32[0x58024400+0x70]))  = 0x8203   (test code code and result4)14、power off openmvPT by SW1（PWR ON/OFF ）      wait one min15、power on openmvPT by SW1（PWR ON/OFF ）16、print(rtc.datetime())  (set test code code and result5)17、print(hex(stm.mem32[0x58024400+0x70]))  == 0x8203    (test code code and result5)

set time codeimport timefrom pyb import RTCrtc = RTC()rtc.datetime((2022, 3, 3, 4, 18, 58, 00, 0))while (True):    print(rtc.datetime())    time.sleep_ms(1000)
result:
(2022, 3, 3, 4, 18, 58, 43, 191406)

0x8103

test codeimport timefrom pyb import RTCimport stmrtc = RTC()#rtc.init()#stm.mem32[0x58024400+0x70]=0x8103print(hex(stm.mem32[0x58024400+0x70]))print(hex(stm.mem32[0x58024800]))while (True):    print(rtc.datetime())    print(hex(stm.mem32[0x58024400+0x70]))    time.sleep_ms(1000)result2:
0x8103

-0xfff3f00

(2022, 3, 3, 4, 18, 58, 38, 335937)

0x8103

(2022, 3, 3, 4, 18, 58, 39, 304687)

0x8103

(2022, 3, 3, 4, 18, 58, 40, 277343)

result3:
0x8203

-0xfff3f00

(2000, 1, 1, 1, 0, 0, 28, 847656)

0x8203

(2000, 1, 1, 1, 0, 0, 29, 808593)

0x8203

(2000, 1, 1, 1, 0, 0, 30, 769531)

0x8203

result4:

0x8203
 
-0xfff3f00
 
(2022, 3, 3, 4, 19, 14, 43, 792968)
 
0x8203
 
(2022, 3, 3, 4, 19, 14, 44, 753906)
 
0x8203

result5:

0x8203
 
-0xfff3f00
 
(2022, 3, 3, 4, 19, 16, 56, 636718)
 
0x8203
 
(2022, 3, 3, 4, 19, 16, 57, 597656)
 
0x8203
 
(2022, 3, 3, 4, 19, 16, 58, 558593)
 
0x8203
 
(2022, 3, 3, 4, 19, 16, 59, 523437)
 
0x8203

weilian1977, 2022-03-04:
now it is ok！

iabdalkader, 2022-03-04:
Can you post the test result ?

weilian1977, 2022-03-04:
computer time   8:57:54,PT time   8:57:09      35  second
computer time 12:43:26,PT time 12:42:35      51  second 

print(hex(stm.mem32[0x58024400+0x70]))=0x8103
print(hex(stm.mem32[0x58024800])) = -0xfff3f00
Now LSE is work,  time can increase with battery connect to PCB board while power off STM32 CPU.But PT time is slower 16 second in about 4 hours。

kwagyeman, 2022-03-04:
Yeah, the LSE probably needs tuning:

https://blog.fearcat.in/a?ID=01600-39c8277a-985b-4709-9205-b459f40e3d36

iabdalkader, 2022-03-08:
@weilian1977 Can you please test this new firmware ? I made a new fix for RTC should work on boards with/without caps.
[openmvpt.zip](https://github.com/openmv/openmv/files/8202392/openmvpt.zip)

weilian1977, 2022-03-08:
can not power on

iabdalkader, 2022-03-09:
Hi, I merged a new fix for boards with/without caps and we tested it, it should be working.. Closing this issue.

closed 2022-03-09 by iabdalkader

weilian1977, 2022-03-10:
where to download the firmware?

iabdalkader, 2022-03-10:
Same link: https://github.com/openmv/openmv/releases/tag/development

weilian1977, 2022-03-14:
Now is OK

iabdalkader, 2022-03-14:
@weilian1977 That's great! Thank you for all the testing.