openmv/openmv #1565: Problems with Cam V3 Kicstarter

aldogent, 2022-03-03:
### Steps to reproduce
1.update firmware
2.erasing and programming
3.play script hello world

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead
After finishing programing the module, the led turns green to red/blue, after pressing play led blinking wihite

### Describe your setup

**IDE version:** 2.9.0

**Firmware version:** 4.2.3

**Operating system:** windows 10

### For developers

**GCC version:**

**Build target:**

iabdalkader, 2022-03-03:
I just tested the M4, both the one that originally shipped with OV2640 and the other one that shipped with OV7725, and they both seem to be working fine. However, the self-test seems to fails due to framebuffer/memory changes that cause the image format to switch bayer. I fix it in the dev release, you can test it if you like download this and upload to the cam (check the erase flash checkbox):
https://github.com/openmv/openmv/releases/download/development/firmware_OPENMV2.zip

aldogent, 2022-03-03:
It's working!!. Thanks a lot. Is this patch only for M4 or you are going to release a new IDE version? Again Thanks!!

iabdalkader, 2022-03-03:
The IDE firmware resources will be updated later.

closed 2022-03-03 by iabdalkader