openmv/openmv #1570: GC2145 image at VGA resolution has washed out colors

sebromero, 2022-03-03:
**Describe the bug**
The camera image shows incorrect colors at VGA resolution. The assumption is that there is something wrong with Bayer to RGB. Could be due to bad bayer sensor registers

**To Reproduce**
Steps to reproduce the behavior:
Set the resolution to VGA with `sensor.set_framesize(sensor.VGA)`
Take a snapshop with `img = sensor.snapshot() `

**Expected behavior**
The colors in the image should look the same as with QVGA resolution.

**Screenshots**
[image]
[image]

**Please complete the following information**
 - IDE version: 2.9
 - Firmware version: 4.2.3
 - Board: NiclaV
 - OS: MacOS

kwagyeman, 2022-03-03:
This is easy to fix. It's just the default bayer pattern setting being wrong. Ibrahim just needs to test the four different patterns we support and it should be fixed.

iabdalkader, 2022-03-03:
I tested them all, this seems to be the right pattern. The issue is this sensor doesn't support processed bayer, it's outputting a raw image (no de-noise, no awb, color correction etc...). I can't find any setting to change it.

closed 2022-03-06 by iabdalkader

iabdalkader, 2022-03-06:
Hey @sebromero  I found some issues with the window and bayer format, the image should look better now.

commit 56a06250 (2022-03-06): sensors: Fix GC2145 window subsampling ratio.

* Fixes #1570
PR #1573 (2022-03-06): sensors: Fix GC2145 window subsampling ratio.
* Fixes #1570