openmv/openmv #1580: First frame has corrupted lines.

iabdalkader, 2022-03-09:
**Describe the bug**
The first frame (technically the second, since we skip the first) shows corrupted lines when setting a window. It's very likely a cache flush/invalidate issue.

**To Reproduce**
```python
import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.set_framebuffers(1)
sensor.set_windowing((0, 0, 240, 240))
sensor.snapshot().flush()
time.sleep_ms(5000)
```

**Screenshots**
[image]

**Please complete the following information**
 - IDE version: [latest]
 - Firmware version: [latest]
 - Board: [OPENMV-4]
 - OS: [Linux]

iabdalkader, 2022-03-09:
CC @kwagyeman

kwagyeman, 2022-04-02:
I can't for the life of me figure out what's causing this. It seems to be highly correlated with sensor.set_windowing((0, 0, 240, 240)). Removing that line removes the issue. It's something to do with windowing being setup. In particular framebuffer->u/framebuffer->v not being equal to the frame size.

kwagyeman, 2022-04-02:
Doesn't seem to have anything to do with jpeg. If I to_png() the image before flush then it still will have the bar. Also, the bar appears to be in the same sport per frame.

kwagyeman, 2022-04-02:
Determined why this happens:

sensor.set_framebuffers(1)
sensor.set_windowing((0, 0, 240, 240))

Doesn't set the frame buffer count to 1. When you do set windowing it resets it back to 3 as the frame buffers auto adjust when you change settings around.

After which, the MDMA writes to the tail image buffer which is different than the head image buffer that's invalidated when snapshot() is called.

So, the solution is invalidate all non-head buffers.

PR coming soon.

kwagyeman, 2022-04-02:
https://github.com/openmv/openmv/pull/1605

iabdalkader, 2022-04-03:
@kwagyeman don't close issues.

kwagyeman, 2022-04-03:
Why? I found the fix?

iabdalkader, 2022-04-03:
@kwagyeman When it's merged the issue will be closed automatically as long as it's linked to a PR. As for the fix, I need to understand it better, I already invalidate buffers before they are returned to the user.

commit 75c80d95 (2022-04-08): imlib: Invalidate vbuffers before returning them.

* Fixes #1580
PR #1605 (2022-05-06): imlib: Fix black lines on first frame.
All frame buffers need to be invalidated before starting MDMA frame capture. The invalidation is done as late as possible so that it does not get repeatedly done over and over again in the sensor_set_*() calls.

I added the method to all MCUs so that if we find we need to do anything else in the future it can be put inside this same method.

Fixes  #1580
closed 2022-05-06 by iabdalkader