openmv/openmv #1584: Nano RP2040 Connect Micropython Thermal Camera Demo: Failed to detect any FIR sensor

jomoengineer, 2022-03-10:
I am attempting to run the Thermal Camera Demo for the Arduino Nano RP2040 Connect From the OpenMV IDE but it is failing to detect the board.  I am using a SparkFun Qwiic GRID-Eye Infrared Array (AMG88xx) board which has an address of 0x69.

This is what the the i2c_scanner.py script shows:

```
Scanning bus 0...
Found device at addres 0:0x60
Found device at addres 0:0x69
Found device at addres 0:0x6a

Scanning bus 1...

```
NOTE: The GRID-Eye device shows as address 0:0x69, or rather 'addres' 0x69. Spelling error?

The error from the thermal_camera.py script shows:
```
Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
ValueError: Failed to detect any FIR sensor
MPY: soft reboot
MicroPython: v1.17-omv-r22 OpenMV: v4.2.3 HAL: v1.3.0 BOARD: Arduino Nano RP2040 Connect-RP2040
Type "help()" for more information.
>>> 
```
[image]

Am I missing something with this example?

IDE version: 2.9.0
Firmware version: 4.2.3
Board: Arduino Nano RP2040 Connect
OS: Windows 10

iabdalkader, 2022-03-10:
The fir sensor needs to be on this bus/pins:

```C
#define FIR_I2C_ID              (0)
#define FIR_I2C_SCL_PIN         (13)
#define FIR_I2C_SDA_PIN         (12)
```

jomoengineer, 2022-03-11:
@iabdalkader 
Thanks again.  I reconnected the GRID-Eye  using the header pins rather than the Qwiic connector and then connected the INT (Interrupt pin) to GPIO0 of the Nano RP2040 and it still does not find the GRID-Eye device.  It still shows as address 0x69 in the i2c_scanner..py output though.  

Note: There is no ID pin on the GRID-Eye device, just an INT Pin.

iabdalkader, 2022-03-11:
You don't need the INT pin, only GND, 3.3V and SCL, SDA connected to A5 and A4. The auto-detection doesn't seem to be working well, so just set the sensor type like this:
```
....
fir.init(fir.FIR_AMG8833)
....
```

jomoengineer, 2022-03-11:
Okay, yeah that worked.

Could this be that the AMG8833_ADDR in the py_fir.c file is set to 0xD2?
```
#define AMG8833_ADDR                    0xD2
```
Thanks.

PR #1587 (2022-03-11): sensors/FIR: Support buses with multiple devices.
* Fixes #1584
commit 3085dc20 (2022-03-11): sensors/FIR: Support buses with multiple devices.

* Fixes #1584
closed 2022-03-11 by iabdalkader