openmv/openmv #1590: Stopping Script in the middle of printing crashes firmware.

kwagyeman, 2022-03-15:
Run this:

```
# Hello World Example
#
# Welcome to the OpenMV IDE! Click on the green run arrow button below to run the script!

import sensor, image, time

sensor.reset()                      # Reset and initialize the sensor.
sensor.set_pixformat(sensor.RGB565) # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QVGA)   # Set frame size to QVGA (320x240)
sensor.skip_frames(time = 2000)     # Wait for settings take effect.
clock = time.clock()                # Create a clock object to track the FPS.

while(True):
    clock.tick()                    # Update the FPS clock.
    #img = sensor.snapshot()         # Take a picture and return the image.
    print(clock.fps())              # Note: OpenMV Cam runs about half as fast when connected
                                    # to the IDE. The FPS should increase once disconnected.

```

Press stop when it starts printing a lot of stuff. On the H7 Plus you crash when this happens. The IDE never gets the USB packet completion from the camera and locks up.

iabdalkader, 2022-03-15:
Found an issue...

iabdalkader, 2022-03-15:
So the issue is, after the IDE issues a STOP command, it stops reading the text buffer, and it becomes full and the printing of the IDE exception blocks for a very long time waiting for something to drain the buffer. The timeout for some reason is 500ms per char, so it eventually breaks which explains why the cam eventually becomes available again to reconnect. Still not sure how to fix it, but working on it.

iabdalkader, 2022-03-15:
So a quick fix is to just reduce the timeout, which will cause the text buffer to overflow, and overwrite the text but the IDE exception will be in the ring buffer. Alternatively, the IDE just needs to drain the text buffer *after* a stop @kwagyeman

kwagyeman, 2022-03-15:
I c CD an just make the IDE keep pooling the text buffer always.

iabdalkader, 2022-03-15:
Either way is fine by me.