openmv/openmv #1596: Portenta Vision Shield LoraWAN AU915

johncaipa, 2022-03-23:
I have seen in the Arduino documentation that the library to support LoraWAN AU915 is not yet ready, I would like to know if someone has already managed to make a working example of this for subband 2, or if someone can tell me what library I could use or check for to help

 - IDE version: 2.9.0
 - Firmware version: ARD-078 1.2.3
 - Board: Portenta H7 and vision shield lora
 - OS: Windows

kwagyeman, 2022-03-23:
https://github.com/openmv/openmv/blob/master/scripts/libraries/lora.py ?

johncaipa, 2022-03-23:
> https://github.com/openmv/openmv/blob/master/scripts/libraries/lora.py ?

I tried the example for Lora with that library and for AU915 it doesn't work

kwagyeman, 2025-02-08:
These problems were resolved on the forums by another user recently: https://forums.openmv.io/t/cannot-register-lora-to-ttn-h7-portenta-vision-shield/10096/18