openmv/openmv #1597: Converting loaded jpegs to RGB565 introduces color shift

kdarras, 2022-03-24:
When loading a jpeg saved on the SD card root, converting it to RGB565 and saving it as JPEG again, the resulting image will have a visible shift in color (less reddish)

Steps to reproduce the behavior:
1. Take full-resolution image(s) with the H7+ board (sensor.RGB565, sensor.snapshot(), img.save("xxx.jpeg",quality=92))
2. Copy jpeg(s) to SD card root
3. Run this:

```
import image, os
if not "output" in os.listdir(): os.mkdir("output")
#scan jpegs on card
files=os.listdir()
jpegs=[files for files in files if "jpg" in files]

for jpeg in jpegs:
    img=image.Image(jpeg,copy_to_fb=True)
    print("original loaded image:",img)
    img.save("straight_"+jpeg)
    
    img.to_rgb565()
    print("rgb565-converted image:",img)
    img.save("torgb_"+jpeg)

```
4. Observe the non-converted ("straight") and rgb565-converted ("torgb565") images

I expected the script and board to output pictures with unaltered colors, whether they were converted to rgb565 or not. Instead, I get visibly different images:

Original image:
[image]

Original image, loaded and then saved again as JPEG:
[image]

Original image, converted to RGB565, and then saved as JPEG:
[image]

 - IDE version: 2.9.0
 - Firmware version: 4.3.0
 - Board: OPENMV-H7+
 - OS: Linux

iabdalkader, 2022-03-29:
@kdarras 

I didn't test this, but note the following

```python
img.save("torgb_"+jpeg)
```
Will compress the image using the default quality which 50%, you can pass a quality arg to `save()`:
```python
img.save("torgb_"+jpeg, quality=92)
```

kdarras, 2022-03-30:
I tried it to be be sure, but it did not make a difference. Even low JPEG quality settings should not introduce systematic color shifts (and note that the non-converted image was also saved with the same settings, and did not have a color shift).

kwagyeman, 2022-04-02:
Hardware jpeg support mainlined: https://github.com/openmv/openmv/pull/1604

iabdalkader, 2022-04-02:
I really think the software decoder should also be fixed, if there's an issue with it, only the H7 has HW JPEG.

kwagyeman, 2022-04-02:
Yes, @bitbank2

bitbank2, 2022-04-02:
I'll look at the code

bitbank2, 2022-04-02:
I just tested this with the attached images on the latest code on the H7+ board. The output loses a little color fidelity, but doesn't turn reddish. Attached are the input and output images I used for testing:

[image]
[image]
[image]

iabdalkader, 2022-04-02:
It could be a rounding issue when converting to RGB565.

kdarras, 2022-04-07:
Hi there, it's about the picture becoming less reddish, @bitbank2 . I just tested it with fresh images (which may need to contain some red to be able to see the shift) on a 4.3.1 firmware H7+ board and the color shift remains.

bitbank2, 2022-04-08:
I found the source of the problem - it's in the JPEGPixelLE and JPEGPixelLE2 methods. The saturation of the color values is not equivalent to the C code I had before with lookup tables. Working on a fix to the SIMD code to avoid adding back 3 x 2K byte tables into FLASH.

bitbank2, 2022-04-08:
Just checked in a fix to my fork:

https://github.com/bitbank2/openmv/commit/981d677e327f68c6b4fe717228bc048d586013f7

iabdalkader, 2022-04-08:
@bitbank2 Does it fix the issue and do you want me to commit it ?

bitbank2, 2022-04-08:
Yes, it fixes the issue

bitbank2, 2022-04-09:
I create a PR
https://github.com/openmv/openmv/pull/1611

PR #1612 (2022-04-09): imlib: Fix SIMD YCbCr to RGB conversion.
* Fixes #1597
commit 5de385ed (2022-04-09): imlib: Fix SIMD YCbCr to RGB conversion.

* Fixes #1597
closed 2022-04-09 by iabdalkader