openmv/openmv #1598: JPEG decoder fails when decoding ROI with to_rgb565

kdarras, 2022-03-25:
When I need to crop jpegs taken by the H7+, the jpeg decoder fails when running to_rgb565 on a roi of the image.

To reproduce:

1. take jpegs with the H7+ board at full resolution, save on SD card root
2. try to crop the jpegs with this script:

```
#import libraries
import sensor, image, time, os, pyb, machine, sys

roi=(0,719,519,1062)

if not "crop" in os.listdir(): os.mkdir("crop")

#scan jpegs on card
files=os.listdir()
jpegs=[files for files in files if "jpg" in files]

#open each jpeg
for jpeg in jpegs:
    print("Loading:",jpeg)
    img=image.Image(jpeg,copy_to_fb=True)

    #check presence of img_s
    try:
        img_s
    except NameError:
        img_s_exists = False
    else:
        img_s_exists = True
    if(not img_s_exists):
        img_s=image.Image(roi[2],roi[3],sensor.RGB565,copy_to_fb=True)
        print("Created empty image with width:",roi[2])
    print("roi:",roi)
    img.to_rgb565(roi=roi,copy=img_s)
    img_s.save("crop/"+jpeg+"_"+str(roi[0])+"_"+str(roi[1])+"_"+str(roi[2])+"_"+str(roi[3])+".jpg")
```

I expect crops of the images to be saved, however, I get: OSerror: JPEG decoder failed
Original images: 
[originals.zip](https://github.com/openmv/openmv-ide/files/8346687/originals.zip)

Note that this script failed differently with the previous firmware 4.2.3, as described here: https://forums.openmv.io/t/load-jpegs-taken-by-h7-for-post-processing/6672/31

IDE: 2.9.0
Board: h7+ with 4.3.0 firmware
Running on Linux

kwagyeman, 2022-04-02:
Hi, the issue with your script is that you are zeroing the frame buffer:

`try:
        img_s
    except NameError:
        img_s_exists = False
    else:
        img_s_exists = True
    if(not img_s_exists):
        img_s=image.Image(roi[2],roi[3],sensor.RGB565,copy_to_fb=True)
        print("Created empty image with width:",roi[2])`

There's only one frame buffer. You loaded a jpeg in there and then you cleared it as a buffer.

Instead do:

`#import libraries
import sensor, image, time, os, pyb, machine, sys

roi=(0,719,519,1062)

if not "crop" in os.listdir(): os.mkdir("crop")

#scan jpegs on card
files=os.listdir()
jpegs=[files for files in files if "jpg" in files]

#open each jpeg
for jpeg in jpegs:
    print("Loading:",jpeg)
    img=image.Image(jpeg,copy_to_fb=True)

    print("roi:",roi)
    img = img.to_rgb565(roi=roi)
    img .save("crop/"+jpeg+"_"+str(roi[0])+"_"+str(roi[1])+"_"+str(roi[2])+"_"+str(roi[3])+".jpg")`

closed 2022-04-02 by kwagyeman

kdarras, 2022-04-07:
Seems to work now, thanks!