openmv/openmv #1599: PureThermal Glitchy Display with fw 4.3.0

griffin175, 2022-03-28:
**Describe the bug**
Glitchy display on every unit flashed with 4.3.0 that I have tested.
6 units tested. 5 out of 6 have had C96 and C97 removed from the RTC crystal.
Happens over HDMI or the onboard display.

**To Reproduce**
1. Flash fw 4.3.0
2. Observe

**To fix**
Flash 4.3.1 (currently the dev release) or the older 4.2.3.

**Screenshots**
Glitch on built in display:
[image]
[image]

Glitch on HDMI output:
[image]
[image]

 - IDE version: IDE 2.9.0
 - Firmware version: fw 4.3.0
 - Board: PureThermal OpenMV
 - OS: flashing from Windows 11

kwagyeman, 2022-04-03:
I managed to cause a wavey lines issue. Then it went away. Different from this... but, signs point to hardware issues.

kwagyeman, 2022-04-03:
This fixes a different issue I was seeing when you'd connect the device to the PC and interrupt the running script and then run it again: https://github.com/openmv/openmv/pull/1607

As for the wavey lines. I need to test more. The wavey line issue isn't easily to reproduce. Still not clear what is causing it.

kwagyeman, 2022-04-03:
As for the above glitch... I used to see that. But, I only get an issue where the display is completely wavey. I don't see the partial image buffer glitch anymore.

iabdalkader, 2022-04-03:
> The wavey line issue isn't easily to reproduce

I think the wavey lines might have something to do with USB storage mounting, I used to see this with normal cams when the host mounts the USB storage and read/write blocks, all IRQs are disabled. Easy way to confirm this is to disable the USB storage.

iabdalkader, 2022-04-25:
I don't see this issue either with the dev-release.

iabdalkader, 2022-05-15:
@griffin175 Is this issue fixed ?

griffin175, 2022-05-31:
@iabdalkader Yes, current release (v4.3.1) is good!

closed 2022-05-31 by griffin175