openmv/openmv #1618: JPEG decoding error when streaming over USB

MasonMcGill, 2022-04-14:
**Describe the bug**
Running a script to stream video over USB using the `pyopenmv` library works for the first two runs, then results in the message "JPEG decode error ()".

**To Reproduce**
- Download `pyopenmv.py`.
- Save the script below to a file called `stream.py` in the same directory (a simplified version of `pyopenmv_fb.py`).
- Run `python3 stream.py` and stop it with an interrupt signal (ctrl+c) three times.

```python
#-- stream.py --#
from time import sleep
import pyopenmv

script = '''
import sensor, image

sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time=2000)

while True:
    sensor.snapshot()
    sensor.flush()
'''

pyopenmv.init(port='/dev/ttyACM0') # Substitute your port name here.
pyopenmv.stop_script()
pyopenmv.enable_fb(True)
pyopenmv.exec_script(script)

try:
    while True:
        if pyopenmv.fb_dump() is None:
            sleep(0.1)
        else:
            print('Got a frame!')
finally:
    pyopenmv.stop_script()
    pyopenmv.disconnect()
```

**Expected behavior**
The third run produces the same behavior as the first.

**Please complete the following information**
 - IDE version: 2.9 (but I'm not using it)
 - Firmware version: 4.3 for OPENMV4 (hopefully I flashed the right version; the firmware naming scheme is a little confusing)
 - Board: OPENMV-H7 R2
 - OS: Linux

MasonMcGill, 2022-04-14:
For context, I'm trying to develop a C module to compile into the firmware, but I'd like to develop/test it on my laptop first, using real data from the OpenMV board, since re-compiling/re-flashing every time I make a small change sounds a little unpleasant. I'm pretty new to OpenMV so it's totally possible I'm going about this the wrong way. Thanks in advance for any help you can provide!

iabdalkader, 2022-04-17:
@MasonMcGill This has nothing to do with JPEG, you need to drain the text buffer if the script generates text output (in this case, the stop() generates an IDE exception log), otherwise the cam will block when the text buffer is full. So basically you just need to add this:
```python
    while True:
        if pyopenmv.fb_dump() is None:
            sleep(0.1)
        else:
            print('Got a frame!')
        # poll and read text buffer
        tx_len = pyopenmv.tx_buf_len()
        if (tx_len):
            print(pyopenmv.tx_buf(tx_len))
```

If there's any connection issues you should follow the _fb.py example when connecting.

MasonMcGill, 2022-04-18:
Yep, that solves it. Good to know the text buffer is something to watch out for. Thank you for your help!

closed 2022-04-18 by MasonMcGill