openmv/openmv #1624: How to run the openmv on raspberry pi pico

qq314541018, 2022-04-20:
I have just compiled the firmware of the target named PICO, and burned the firmware.uf2 to the pico board , but it seems can not run.
How can I resolve it？Thanks for you reply.

iabdalkader, 2022-04-20:
That should have worked, I don't have a PICO with me now to test with, try disabling things in main.c and see where it's getting stuck.

qq314541018, 2022-04-20:
> That should have worked, I don't have a PICO with me now to test with, try disabling things in main.c and see where it's getting stuck.

When I connect the pico to the openmvide, always the openmvide pops up a dialog box of updating the firmware. And I can not do anything else.

sgbaird, 2024-05-23:
xref: https://forums.openmv.io/t/trouble-running-rpc-functions-on-raspberry-pi-pico/7579/2