openmv/openmv #1625: Shallow clone instructions don't work with Git 2.25

MasonMcGill, 2022-04-23:
**Describe the bug**
Passing `--no-single-branch` to `git submodule update` as described in https://github.com/openmv/openmv/blob/master/src/README.md confuses my version of Git. The source download/firmware compilation process seems to work fine without the flag, but I'm deeply ignorant of the Git submodule system, so it's totally possible some features are missing. Doing a fully recursive clone works fine on my system (but takes a while).

**To Reproduce**
```sh
git clone --depth=1 https://github.com/openmv/openmv.git
cd openmv
git submodule update --init --depth=1 --no-single-branch
git -C src/micropython/ submodule update --init --depth=1
```

**Expected behavior**
The source files needed for the firmware download successfully.

**Screenshots**
[image]

**Please complete the following information**
 - IDE version: N/A
 - Firmware version: Master (d10c883e6557bd62bc105388360542407094ecdb)
 - Board: OPENMV-H7 R2
 - OS: Ubuntu 20.04

iabdalkader, 2022-04-24:
That option  `[--[no-]single-branch]` has been available since git [2.8.6](https://git-scm.com/docs/git-clone/2.8.6), and that same command line is used in the firmware build workflow (which runs on Ubuntu 20.04): 

https://github.com/openmv/openmv/blob/master/.github/workflows/firmware.yml#L28
https://github.com/openmv/openmv/blob/master/tools/ci.sh#L24

I think it's a local issue, maybe you have a bash alias for git.

MasonMcGill, 2022-04-24:
Hmm; I think I just have the standard Git you get running `apt install`:

[image]
 
And uninstalling & reinstalling didn't change anything. This isn't really an issue for me personally, since I don't anticipate needing to clone this repository very frequently, so if it's just affecting me I wouldn't worry. Very mysterious though!

iabdalkader, 2022-04-24:
Check `alias` ? just type alias in the command line.

MasonMcGill, 2022-04-24:
[image]

MasonMcGill, 2022-04-24:
Here's my git config:
```
user.email=mmcgill@caltech.edu
user.name=MasonMcGill
core.repositoryformatversion=0
core.filemode=true
core.bare=false
core.logallrefupdates=true
remote.origin.url=https://github.com/openmv/openmv.git
remote.origin.fetch=+refs/heads/master:refs/remotes/origin/master
branch.master.remote=origin
branch.master.merge=refs/heads/master
```

I'm not much of a Git expert so I don't have a sense of if there's anything funny in there.

MasonMcGill, 2022-04-24:
Woops; that's from the `openmv` directory. Here's what `git config --list` prints out in my home directory:
```
user.email=mmcgill@caltech.edu
user.name=MasonMcGill
```

iabdalkader, 2022-04-24:
I just realized I was looking at the wrong docs. It seems `[--[no-]single-branch]` was actually added in [2.26.0](https://git-scm.com/docs/git-submodule/2.26.0) for git-submodule. The github workflow Ubuntu-20.04 VM runs a more recent git (2.35.1) from repository: ppa:git-core/ppa. To fix this you need to run at least 2.26.0, or just remove the flag.

MasonMcGill, 2022-04-24:
Oh ok that makes sense. Thanks for looking into it!

closed 2022-04-24 by iabdalkader